microscope-cockpit/cockpit #504: executorDevice hard codes number of analog and digital channels

carandraug, 2019-06-10:
The executor handlers take an argument for the number of digital and analogue line. However, `ExecutorDevice` has an hardcoded value of 16 digital and 4 analogue lines. Our current real life case for this issue is that the RedPitaya only has 2 analogue lines. We can't ignore the missing lines because preparing the action table requires reading the values on all lines (`ExecutorHandler.executeTable` will read all analogue values at the start).

I guess we could have a cockpit device for each. However, I think maybe it would be better to query the remote for the number of digital and analogue lines? If the later, then this is also an issue for microscope.

`LegacyDSP` also hardcodes the number of lines but I guess that's not really an issue.

juliomateoslangerak, 2019-06-10:
Would it be an option to have Nr of analogues and digitals in the configuration ?

carandraug, 2019-06-10:
@juliomateoslangerak  wrote:
> Would it be an option to have Nr of analogues and digitals in the configuration ?

Oh yeah, that's also another option.

However, I think the remote should know the number of line it has and there should be no need for configuration.

mickp, 2019-07-01:
Fixed in 7539061010150043336a2fd2aed717a4c392b882 for new executors.

LegacyDSP still uses hard coded values, because they're fixed for this hardware. Also, it duplicates the all but two lines of the getHandlers method where these values are hard coded. We can probably remove this duplicated method, but LegacyDSP.getAnalog and LegacyDSP.setAnalog need to be tested with the hardware. I don't think they're ever used, but would be if we used the base class' getHandlers, and probably need updating anyway.
 * LegacyDSP.setAnalog caches the output value before setting it wit MoveAbsoluteADU. I think it should now use setAnalog.
 * LegacyDSP.getAnalog returns the cached values, because the old DSP remote had no way to correctly read back the current analogue output (there were a number methods that looked like they would read it back, but they each seemed to apply different, undocumented and arbitrary scaling factors). The new, cleaner C wrapper may do a better job, but needs testing.

mickp, 2019-07-01:
Right - the handler for LegacyDSP is *already* using the new remote functions, and they work fine. That means I can just point LegacyDSP.setAnalog and .getAnalog to those remote functions, then ditch the duplicated getHandlers.

iandobbie, 2023-08-22:
This is fixed for the generic executor device and just hanging around for the legacyDSP device which has values fixed for the hardware we use. I am closing this as fixed

closed 2023-08-22 by iandobbie