microscope-cockpit/cockpit #506: during experiment cleanup, stage position is reset twice

carandraug, 2019-06-11:
At the end of an experiment, during `experiment.cleanup()` the z height is reset twice:

1. `experiment.cleanup` publishes "cleanup after experiment". `ExecutorHandler` is subscribed to it, eventually calling `AnalogMixin._cleanupAfterExperiment` which restores the position to the value before experiment.
2. `experiment.cleanup` then also resets back to the z altitude it saved before the experiment.

Is there a reason to have it done twice (a case where only one of them is triggered) or can one of them be removed?

mickp, 2019-06-13:
No, there is no reason to reset the stage position twice at the end of an experiment. We should choose or implement one method which will reset the position in all these cases:
 * experiment execution completes normally;
 * experiment execution is aborted by the user;
 * experiment execution fails due to an exception.

carandraug, 2019-06-25:
Failing an exception during `experiment.cleanup` both current methods cover those three cases.

There is one other case that none of the current methods covers. The cleanup thread is only setup at the end of the `experiment.run` but the Z stage was moved before, during `experiment.prepareHandlers`. If an exception happens during that preparation, the system will be left in an unknown state.

Anyway, addressing the current issue of duplicated cleanup, I think having the experiment class explicitly resetting the z position is clearer than having the Executor doing it i via the events system. I would propose we remove the `prepareForExperiment` and `cleanupAfterExperiment` from the executr handler then.

Looking forward, I think we could use some sort of [memento pattern](https://en.wikipedia.org/wiki/Memento_pattern) to address this. The experiment class would create a list of mementos during its preparation stage. Whatever is waiting for the experiment to end is then responsible for rolling back those mementos.  This should cover failure in any case and would work for any device setting.