microscope-cockpit/cockpit #507: experiment with `executorDevice` crashes if longer than 6 seconds

carandraug, 2019-06-11:
The `executorDevice` connection to the remote has an hardcoded timeout value of 6 seconds. This causes problems during experiment because we then call `RunActions` on the remote and wait for it to return:

```
    def executeTable(self, table, startIndex, stopIndex, numReps, repDuration):
...
        events.executeAndWaitFor(events.EXECUTOR_DONE % self.name,
                                 self.connection.RunActions)
```

If the experiment is longer than 6 seconds, then this errors with a Pyro4 timeout error.

A related issue is: after the timeout error, the experiment triggers a cleanup and all appears fine. However, the remote continues to move the stage and trigger the devices as if nothing happened. I guess that as part of cleanup, experiment should ensure that the executor really stops whatever it may be doing?

carandraug, 2019-06-11:
I spoke with @mickp  about the issue which remembers the timeout being there because the old DSP was constantly crashing. There should have been code somewhere that would remove the timeout while running the experiment and restore it afterwards.

iandobbie, 2020-10-22:
The issue seems to be more involved than the above explanation. On the dsp (eg diamond cryosim) experiments of several minute and 1000's of exposures work fine.  Howeverm, on the Red Pitaya long experiments seem to generate a timeout error. Nick experiences this with long action tables, but doesn't seem to be dependant on the exposure time. His SIM experiments with 2 channels over about 15 um seem to crash. 

This would be 15 images per z, 8 z's per micron, 2 channels, so about
 15*8*2*15 = 3600 images. 
]Each image needs 2 lines for camera, 2 lines for laser 2 lines for pol rotator, 2 lines for SLM. so 
3600*8= 28,800 lines. 
Additionally each z needs 2 entries, so an additional 240 lines, so about 30,000 lines.

iandobbie, 2021-04-19:
I suggest something along the lines of:

        self.connection.PrepareActions(actions, numReps)
        #store cuirrent timeout and reset to infinite for experiment.
        origTimeout = self.connection._proxyTimreout
        self.connection._pyroTimeout = 0
        events.executeAndWaitFor(events.EXECUTOR_DONE % self.name, self.connection.RunActions)
        self.connection._pyroTimeout = origTimeout
        events.publish(events.EXPERIMENT_EXECUTION)
        return

I will try it next time I am on DeepSIM. 

I wonder if the difference arrises because the DSP returns once the experiment starts as it is all running on the DSP card. Maybe we just need to make the Red Pitaya do the same thing, then we wouldn't need to do this.

iandobbie, 2021-04-19:
> The issue seems to be more involved than the above explanation. On the dsp (eg diamond cryosim) experiments of several minute and 1000's of exposures work fine. Howeverm, on the Red Pitaya long [...]

I think this is a different issue, related to the camera triggers being missed covered in microscope issue https://github.com/python-microscope/microscope/issues/203

iandobbie, 2021-04-19:
I think that the DSP returned as soon as the collection was started whereas the red pitaya blocks, which causes this issue. The dsp code runs this when the experiment is started:-

    def trigCollect(self):
        # If trigCollect is called without first downloading and initing a profile,
        # this process dies abruptly (presumably hitting some access violation in 
        # the low-level code?). Don't let this happen.
        if not self._profileInited:
            raise Exception("Need to InitProfile before trigCollect.")
        self.collThread.do('collect', uid=uuid.uuid1())
        self._profileInited = False

(code from https://github.com/mickp/cockpit-dspremote/blob/master/dsp.py)

Does this return immediately? I'm not sure.

iandobbie, 2021-04-22:
The fix suggested above does not work as it causes the whole front end to lock up. We need the Red Pitaya to return as soon as the experiment is started. The system then should wait for the experiment complete signal from the red pitaya. I am not sure why this appears to work on the zaber setup. I will investigate.

iandobbie, 2021-04-26:
I have a fix on the red pitaya on the zaber but was having other, I think unrelated issues. Effectively it started the action table in a new thread and returned as soon as the setup and start was performed. This completely solved the timeout issues, but I was still struggling intermittently with experiments but I think this was a ximea camera issue. I am going to investigate further and make sure we have a copy of the corrected code in the red pitaya repository.

iandobbie, 2021-04-27:
Both the zaber and DeepSIM are working with the fix on the red pitaya. I just changes the RunActions function to start a new thread and return as soon as the experiment is started.

    def RunActions(self):
        runThread=threading.Thread(target=self.trigCollect)
        runThread.start()
        return

The git repository on the red pitaya is massively updated from its original git repository. I will  copy the code from my not networked RP and synch the repositories.

iandobbie, 2021-04-27:
Pushed the current running red pitaya dsp code to https://github.com/MicronOxford/RedPitaya-DSP/tree/ian-working.

This incorporates the 8 to 16 digital line fix and the timout issue here.