microscope-cockpit/cockpit #510: Using the light path option does not refresh the cameras status buttons

juliomateoslangerak, 2019-06-12:
I'm using the new GUI rework on a mac. All in dummy mode
When I use the light path button option to switch between one or two cameras, the buttons do not refresh to update the camera status.
The light source does not seem to update either, but that might be intended...

mickp, 2019-06-13:
Ian will need to comment on this: it's not entirely clear to me what the path buttons should and should not set - I thought it was just flip mirror states and objective.

iandobbie, 2019-06-18:
I'm sorry I dont quite understand what you are wanting to do. In all dummy mode there aren't any light path mode buttons are there? I dont see any.

However the very latest openGL changes do seem to kill it on my machine, unfortunately I also just updated my OS, so I have no idea which killed it. I will investigate

juliomateoslangerak, 2019-06-19:
I see this in a dummy mode in my mac laptop. I create a few microscope dummy lasers, filters and cameras.
Using the Light path button I create two new paths, One with two cameras and another with 3 cameras. If I laters swap from 3 to two cameras using the created light paths, the camera deactivates in the camera view canvas but the camera toggle button remains active. I can upload a screenshot later. 
So independently of the use, that seems to be a bug. Maybe never noticed if it was never used with this intent...

Julio

mickp, 2019-06-19:
AFAIK, cameras, filters and lights do not constitute a path; paths are
determined by the state of flip mirrors and the selected objective. Either
way, something is wrong here, but it will not be possible to fix it until
the intended behaviour of these buttons is clearly defined.

juliomateoslangerak, 2019-07-23:
My view on the light path was to have a way for the user to quickly save and change configurations, including cameras, lights, etc... But that would be a bit redundant with the save/load experiment settings... though that is in the experiment setup window...
What is the use case of the light path button?
Would it make sense for you to have a save/load setting button in the main window?

carandraug, 2020-05-08:
I'm looking into the light path stuff now and I can't reproduce this issue. Starting cockpit with `--no-config-files`, I load these two light path configurations:

```
# enabled camera 1 and 2, and lights 405 and 488
$ cat ~/MUI_DATA/carandraug/foo.txt 
{"objective": "40x", "Dummy camera 1": true, "Dummy camera 2": true, "Dummy camera 3": false, "Dummy camera 4": false, "Dummy 405 power": 0.01, "Dummy 405": {"isEnabled": true, "exposureTime": 100}, "Dummy 488 power": 0.01, "Dummy 488": {"isEnabled": true, "exposureTime": 100}, "Dummy 633 power": 0.01, "Dummy 633": {"isEnabled": false, "exposureTime": 100}}

# enabled camera 2 and 3, and lights 488 and 633
$ cat ~/MUI_DATA/carandraug/bar.txt 
{"objective": "40x", "Dummy camera 1": false, "Dummy camera 2": true, "Dummy camera 3": true, "Dummy camera 4": false, "Dummy 405 power": 0.01, "Dummy 405": {"isEnabled": false, "exposureTime": 100}, "Dummy 488 power": 0.01, "Dummy 488": {"isEnabled": true, "exposureTime": 100}, "Dummy 633 power": 0.01, "Dummy 633": {"isEnabled": true, "exposureTime": 100}}
```

I can switch between the two and see that both light sources and cameras enable/disable/remain appropriately and that it shows correctly on the main window UI.

@juliomateoslangerak can you confirm if you still have this issue? If so, can you give some steps to reproduce it?

iandobbie, 2020-05-11:
I did a lot of fixing of this functionality so hopefully it is now all working.

carandraug, 2020-05-12:
I'm also pretty confident the issue does not exist anymore, so I'm closing as fixed. @juliomateoslangerak please comment here to reopen the issue if you can still reproduce your issue.

closed 2020-05-12 by carandraug