microscope-cockpit/cockpit #511: macos 10.14 breaks cockpit.

iandobbie, 2019-06-19:
Just upgraded to macos 10.14 and am running into wx issues. Here is an example but there are several similar error saying  we call wxPaintDC wihtout being in a paint event, but the code doesn't seem to be doing this. I think something else must be the core of the issue. I will try to chase it up.

Traceback (most recent call last):
  File “/Users/ian/src/cockpit/cockpit/gui/mosaic/canvas.py”, line 348, in onPaint
    dc = wx.PaintDC(self)
wx._core.wxAssertionError: C++ assertion “window->MacGetCGContextRef() != NULL” failed at /Users/robind/projects/buildbots/macosx-vm6/dist-osx-py37/Phoenix/ext/wxWidgets/src/osx/carbon/dcclient.cpp(195) in wxPaintDCImpl(): using wxPaintDC without being in a native paint event

mickp, 2019-06-19:
The mosaic code is shared by the mosaic and the touchscreen, but each
should create its own DC. Do you still see the error if you eliminate the
touchscreen code (i.e. comment out touchscreen-related lines 181 and 182 in
```cockpit/__init__.py```)?

iandobbie, 2019-06-20:
Removing the touchscreen doesn't help. Updating the mosaic canvas onPaint call to return if the GLinit functions haven't been called fixes the first iteration of this, however the macrostage Z module has two classes, MacroStageZ and MacroStageZKey which both fail in the onPaint function in  the same way. It is strange they seem to work correctly first time they are called but fail the second time. I am still investigating.

mickp, 2019-06-21:
I vaguely remember that when I rewrote the GL stuff for the viewCanvas, on the Mac, there was sometimes a problem where creating the required (but unused) PaintDC instance. I fixed it by replacing:
```
    dc = wx.PaintDC(self)
```
with
``` 
    try:
        # Unused, but wx requires we create an instance of PaintDC.
        dc = wx.PaintDC(self)
    except:
        return
```
in onPaint.

iandobbie, 2019-06-21:
This removes the error but we now don't have all the graphics canvases. It might be some of the other changes I have made. I will investigate and start from master again.

mickp, 2019-07-08:
027499c50e013be470d46ce5347c86f085eb6566 fixes one issue that prevented the touchscreen initialising correctly.

There is still an issue with every glCanvas instance. I think it's related to problems creating the PaintDC for handling their paint events, which may be a wxPython bug. If I don't create the wxPaintDC in the EVT_PAINT handler, there are no more errors, but windows containing a glCanvas are only updated when there is some other event (e.g. a mouse event) after the glCanvas paint event ... this is probably because the wxPaintDC keeps track of regions that need to be redrawn.

I've opened a wxWidgets issue at wxWidgets/phoenix#1300 .

iandobbie, 2019-08-31:
As described by Mick this is a wxWidgets issue, which requires an updated wxwidgets and wxpythoin (as ti has its own copy of wxWidgets). The issue is fixed by 
https://github.com/wxWidgets/wxWidgets/commit/531fdbcb64b265e6f24f1f0cc7469f308b9fb697

I have compiled this and built wxPython which fixes the core issue. I will build a wheel and show it on micronadmin.

mickp, 2019-09-05:
I tried the wheel. I don't see the error any more, but canvases are only updated if there is some other event on their window.

mickp, 2019-09-05:
The update issue was a threading issue. d2a8349c48ce11bdc6aabd37589357737e78b0cc fixes it for ViewCanvas. I'll take a look at the mosaic now.

mickp, 2019-09-05:
Mosaic updates fixed in 433e3ef48d438127e6a8236a33404f79377aef02.

closed 2019-09-05 by mickp