microscope-cockpit/cockpit #514: Single Site experment not working

asantentata, 2019-06-26:
When single site experiment is run the following standard error appears on the logging panel:

```
Traceback (most recent call last):
  File "c:\cockpit\cockpit\gui\dialogs\experiment\singleSiteExperiment.py", line 130, in onStart
    if self.panel.runExperiment():
  File "c:\cockpit\cockpit\gui\dialogs\experiment\experimentConfigPanel.py", line 519, in runExperiment
    return self.runner.run()
  File "c:\cockpit\cockpit\experiment\experiment.py", line 262, in run
    cameraToExcitation)
  File "c:\cockpit\cockpit\experiment\dataSaver.py", line 132, in __init__
    width, height = camera.getImageSize()
  File "c:\cockpit\cockpit\handlers\camera.py", line 212, in getImageSize
    return self.callbacks['getImageSize'](self.name)
  File "c:\cockpit\cockpit\devices\microscopeCamera.py", line 256, in getImageSize
    binning = Binning(binning)
TypeError: __new__() missing 1 required positional argument: 'v'
Exception in thread Experiment-execute:
Traceback (most recent call last):
  File "C:\Program Files\Python37\lib\threading.py", line 917, in _bootstrap_inner
    self.run()
  File "C:\Program Files\Python37\lib\threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "c:\cockpit\cockpit\experiment\experiment.py", line 397, in execute
    curIndex + bestLen, numReps, repDuration)
  File "c:\cockpit\cockpit\events.py", line 187, in executeAndWaitFor
    func(*args, **kwargs)
  File "c:\cockpit\cockpit\handlers\executor.py", line 207, in executeTable
    repDuration)
  File "c:\cockpit\cockpit\devices\executorDevices.py", line 193, in executeTable
    self.connection.PrepareActions(actions, numReps)
  File "C:\Program Files\Python37\lib\site-packages\Pyro4\core.py", line 185, in __call__
    return self.__send(self.__name, args, kwargs)
  File "C:\Program Files\Python37\lib\site-packages\Pyro4\core.py", line 476, in _pyroInvoke
    raise data  # if you see this in your traceback, you should probably inspect the remote traceback as well
IndexError: list index out of range
```

mickp, 2019-06-26:
Should be fixed by f52768d3638baeae8c04092f87873d0c87171efc. Please checkout lastest and test.

asantentata, 2019-06-26:
It did not work. We have new error message as below.

Exception in thread Experiment-execute:
Traceback (most recent call last):
  File "C:\Program Files\Python37\lib\threading.py", line 917, in _bootstrap_inner
    self.run()
  File "C:\Program Files\Python37\lib\threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "c:\cockpit\cockpit\experiment\experiment.py", line 397, in execute
    curIndex + bestLen, numReps, repDuration)
  File "c:\cockpit\cockpit\events.py", line 187, in executeAndWaitFor
    func(*args, **kwargs)
  File "c:\cockpit\cockpit\handlers\executor.py", line 152, in executeTable
    astate = [self.getAnalogLine(line) for line in range(self._alines)]
  File "c:\cockpit\cockpit\handlers\executor.py", line 152, in <listcomp>
    astate = [self.getAnalogLine(line) for line in range(self._alines)]
  File "c:\cockpit\cockpit\handlers\executor.py", line 385, in getAnalogLine
    return self.callbacks['getAnalog'](line)
  File "C:\Program Files\Python37\lib\site-packages\Pyro4\core.py", line 185, in __call__
    return self.__send(self.__name, args, kwargs)
  File "C:\Program Files\Python37\lib\site-packages\Pyro4\core.py", line 476, in _pyroInvoke
    raise data  # if you see this in your traceback, you should probably inspect the remote traceback as well
Exception: Unexpected analog channel! (0 or 1)

mickp, 2019-06-26:
The original problem is fixed. The second error message is due to issue #504 .

iandobbie, 2019-06-26:
I think the Ximea camera driver isn’t returning a proper roi. I started to work on it but didn’t have a camera at home to test. I will check it out

Ian

mickp, 2019-07-01:
The second error message should be fixed in 7539061010150043336a2fd2aed717a4c392b882, if you add an 'alines' and 'dlines' entry to the RedPitaya's config section, specifying the number of available analogue and digital lines, respectively.

asantentata, 2019-07-08:
After adding alines an dlines into dsp section on config file, we get the following errror in Cockpit:

Exception in thread Experiment-execute:
Traceback (most recent call last):
  File "C:\Program Files\Python37\lib\threading.py", line 917, in _bootstrap_inner
    self.run()
  File "C:\Program Files\Python37\lib\threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "c:\cockpit\cockpit\experiment\experiment.py", line 397, in execute
    curIndex + bestLen, numReps, repDuration)
  File "c:\cockpit\cockpit\events.py", line 187, in executeAndWaitFor
    func(*args, **kwargs)
  File "c:\cockpit\cockpit\handlers\executor.py", line 152, in executeTable
    astate = [self.getAnalogLine(line) for line in range(self._alines)]
  File "c:\cockpit\cockpit\handlers\executor.py", line 152, in <listcomp>
    astate = [self.getAnalogLine(line) for line in range(self._alines)]
  File "c:\cockpit\cockpit\handlers\executor.py", line 385, in getAnalogLine
    return self.callbacks['getAnalog'](line)
  File "C:\Program Files\Python37\lib\site-packages\Pyro4\core.py", line 185, in __call__
    return self.__send(self.__name, args, kwargs)
  File "C:\Program Files\Python37\lib\site-packages\Pyro4\core.py", line 476, in _pyroInvoke
    raise data  # if you see this in your traceback, you should probably inspect the remote traceback as well
Exception: Unexpected analog channel! (0 or 1)

closed 2019-10-09 by carandraug