microscope-cockpit/cockpit #515: camera views with merged channels

carandraug, 2019-07-02:
The camera views window displays the images as acquired. If two cameras are enabled, they are always displayed side-by-side. Would be nice if it was possible to display the two channels merged somehow.

juliomateoslangerak, 2019-07-17:
I believe one of the strengths of cockpit is showing the images separated. 
I can imagine that a lot of users would ask for this.

iandobbie, 2019-10-23:
If you really have 2 cameras then merged images are not what you want as they wont be aligned. I think we should just close this and leave it. Maybe the view last image window could get colour and merged images but I don't think so.

juliomateoslangerak, 2019-10-23:
I could imagine cases where there are not really two physical cameras. It could be a splitter projecting two channels into the same camera (maybe better alignment).
I still think a lambda user will want this feature.
One possible alternative could be a synced view of the multiple cameras. Very much like the twin slicer of Huygens. Zoom and panning are in sync and also a dynamic autoscale of the viewed area. It’s great to explore the details without the need of messing with the scaling back and forth

iandobbie, 2022-05-22:
I have added the functionality to toggle a flag on any camera view. If you activate this then motions in other windows, eg pan or zoom will update the view in that window. I have called this locked views but maybe better refered to as synced windows. I guess I should go change my event names and menu items but the functionality works.

iandobbie, 2022-05-22:
Changed names to synced views which is a better description of what this does.

closed 2022-05-22 by iandobbie