microscope-cockpit/cockpit #522: Experiment gets wrong camera_ready timings

juliomateoslangerak, 2019-07-11:
Experiment uses getTimeBetweenExposures to calculate when the camera is ready for a next exposure. It should rather use the camera readout_time.
getTimeBetweenExposures is returning the inverse of the camera acquisition rate (at least with andors sdk3), which depends on the exposure time on the camera

mickp, 2019-07-11:
Then it's wrong in the sdk3 implementation.  From handlers\camera.py:

```python
    ## Return the amount of time, in milliseconds, that must pass after
    # ending one exposure before another can be started.
    # If isExact is specified, then we return a decimal.Decimal value instead
    # of a raw floating point value.
    @cached
    def getTimeBetweenExposures(self, isExact = False):
        return self.callbacks['getTimeBetweenExposures'](self.name, isExact)

juliomateoslangerak, 2019-07-11:
get_cycle_time returns `1. / self._frame_rate.get_value()`. Subtracting the exposure time should work but maybe prone for errors. We should then subtract the exposure time internal to the camera.
Would not be just more convenient to provide a get_readout_time?

mickp, 2019-07-11:
I chose cycle time because that's the one thing we can definitely figure out for any camera.

* Some cameras give minimum cycle time (exposure + readout) directly.
* Some cameras give info about readout time and exposure time, so we just add them;
* Some cameras only give a max framerate, so we take the reciprocal.
* Some cameras which work on a framerate basis only give control over the exposure time through manipulation of the framerate - you've no idea what the real exposure time is, unless they also give you info about readout time.

Adding parameters to determine a cycle time on one machine then subtracting them on another is problematic, as it introduces floating point error.

We use the cycle time to ensure that triggers aren't sent faster than the camera can deal with them. We shouldn't really need the exposure time unless the camera only offers bulb-type exposure: otherwise, it should be something we configure (either directly, or doing the best we can via framerate), then only care about for metadata.

There are couple of distinct cases we need to handle. For cameras that trigger on an edge or level, we should ensure that experiments use the cycle time to test against (and increase, as necessary) the time between those points in consecutive triggers in the action table.

The other case is bulb exposure. There, we have to use the readout time (or our best guess) to test against/increase the time between the trigger signal's falling edge and its next rising edge.

juliomateoslangerak, 2019-07-17:
The use case I'm trying to deal with is the last one, bulb exposure. I'm trying to get the andor zyla to acquire z stacks as quick as possible. 
The Zyla 4.2 plus only has a rolling shutter. So I want it to trigger the lasers only when all rows are exposed to avoid the smearing effect. That is what I refer as pseudoglobal in issue #441. 
So the experiment has to figure out the read_out_time and, first, trigger the cameras that much in advance and, second, add that figure to calculate the camera_ready_time.
I would think that should be the default behaviour in rolling shutter cases... Do you imagine cases where it would not?
That's why it makes sense for me to know the camera_readout_time, as subtracting the exposure_time in the experiment would be fishy. I have to be sure that the camera also has the same exposure time that is configured in my experiment.
Whose responsibility is to do all this logic? The experiment or the camera interface. Would it make sense to implement a get_readout_time and that the micrsocopeCamera decides what is best and fall back to cycle_time if it cannot be calculated?

juliomateoslangerak, 2019-07-23:
Actually @iandobbie fixed this in #509.

closed 2019-07-24 by juliomateoslangerak