microscope-cockpit/cockpit #523: No Stage kills the mosaic

iandobbie, 2019-07-15:
If you don't have a functioning stage (eg Linkam stage is powered off at Diamond) the mosaic behaves very badly. We need to trap this condition in some manner and at least warn the user.

mickp, 2019-07-16:
The issue isn't that there's no stage --- then we fall back to a dummy stage device --- but that a Linkam stage is specified in the config yet is not connected.  The cockpit-side ```linkamStage``` module includes ```DEFAULT_LIMITS``` to fall back to in the event of an exception when querying the stage's motor limits. However, these are not used, as querying the motor limits when the stage is disconnected just returns limits of (0,0), rather than throwing an exception. We can fix this by testing for zero-valued limits.

Looking into this exposed some other issues with hard limits, though.

mickp, 2019-07-16:
Fixed in 34432197eea5419dbc1e17cb968343c39a6ab9aa.

(Will open a new issue for the other hard limits stuff.)

closed 2019-07-16 by mickp