microscope-cockpit/cockpit #524: hardLimits look (and used as if) dynamic, but resolves to constants.

mickp, 2019-07-16:
Currently, ```interfaces.stageMover.getHardLimits``` queries ```getHardLimits``` on each ```stagePositioner.PositionerHandler```. The base class simply returns a tuple that was populated during it's ```__init__```, so ```getHardLimits``` effectively returns a constant.

Making hardLimits truly dynamic looks like a lot of work.

```MosaicWindow.__init__``` queries ```interfaces.stageMover```'s hard limits and uses them to create its canvas, which in turn uses them to determine how many ```MegaTile```s it needs. There is no existing way to update this later, and it is sufficiently complicated that to do so would require a significant rewrite of the Mosaic code.

```TouchScreenWindow.__init___``` queries ```interfaces.stageMover```'s hard limits, and uses them to create its canvas in the same way as ```MosaicWindow```, with the same issues.

```MacroStageXY.__init__``` queries ```interfaces.stageMover```'s hard limits and uses them to determine its bounds and extent. These are stored as attributes, so it may be possible to update these after creation, but care must be taken to ensure that all derived attributes are updated accordingly. This would require refactoring ```MacroStageXY``` to put all the derived attribute calculations in one function, rather than determining them *ad hoc* throughout the code.

```MacroStageXY``` also queries the hard limits in its ```EVT_PAINT``` handler (twice) - will raise a separate issue for this.

carandraug, 2020-05-21:
> Making hardLimits truly dynamic looks like a lot of work.

Do we want this? I would imagine that a stage hard limits don't change, it's the soft limits that need to be dynamic. In that case, this issue is about changing the code to be more efficient and not query the hard limits so many times.

Or do we need dynamic hard limits?

iandobbie, 2020-05-22:
I think hardlimits are device or instrument specific so reading them from hardware or config files once is the correct approach. If they need to change having to edit a config file and restart the code is acceptable.

carandraug, 2020-05-22:
In that case, I have a proposed fix on my `524-compute-hard-limits-once` branch. It replaces the body of `stageMover.getHardLimits` and `stageMover.getHardLimitsForAxis` functions with getters to the `StageMover` new `hard_limits` attribute which is computed during construction.

mickp, 2020-05-22:
I never really understood the use of this term in cockpit: by definition,
there are only soft limits in the software; hard limits are hard stops or
limit switches on the mechanical assemblies. I'm not even sure what
mechanism there is to go beyond the soft limits to the hard limits.

carandraug, 2020-05-22:
> I never really understood the use of this term in cockpit: by definition,
> there are only soft limits in the software; hard limits are hard stops or
> limit switches on the mechanical assemblies.

My understanding is that hard limits in software are a representation of the mechanical limits and so cannot be changed (which is why my proposal for this issue is to not make them dynamic). Soft limits can be changed over time by the user, and the soft limits are limited by the value of hard limits. In a way, the main value of hard limits is to prevent a user from setting invalid soft limits, as well as a default value for the soft limits.

But that's only what I have always assumed to be, I never asked anyone, so maybe I'm wrong.

> [...] I'm not even sure what
> mechanism there is to go beyond the soft limits to the hard limits.

I think the whole point is that there shouldn't be a mechanism (in software) to go beyond the soft limits (other than changing them).

carandraug, 2020-06-01:
I don't think @mickp comment was against the proposed change, or was it? If not, I would like to merge this. @iandobbie do you have anything to add on what does soft vs hard limits mean in cockpit?

mickp, 2020-06-02:
No objection here - more an observation that, IMO, the hard limits serve little purpose, so could probably be done away with,

iandobbie, 2020-06-02:
I was going to suggest the opposite, get rid of the soft limits. I think the concept of hard limits is a good idea as either the hardware might not have sufficent limit protection (I know crappy design but I'm sure you can imaging this) or more likely the system as a whole is unable to utilise the whole of the hardware range of the stage and so we want to avoid moving beyond a set limit. 

My understanding of the use case is the hardlimits are the the physical range over which the stage can move, with soft limits being set per sample, say the edge of a coverslip, or the end of the sample. In my experience I have never used the soft limits as its is just an additional step in setup which I haven't bothered with. Navigating the sample I do soft limit checks by eye. As each sample isn't on the microscope that long  it doesn't seem worth setting the soft limits.

Back to the main topic, making the hard limits constants is a good idea, if they need chnaging and software restart is not unreasonable.

mickp, 2020-06-02:
We're talking about the same thing. My take was that a hard limit is in hardware (a mechanical stop or cutout switch); if it's in software, it's a soft limit by definition.

carandraug, 2020-06-02:
Seems like we are all in agreement with the fix for the original issue. I have rebased and merged the proposed fix above, i.e., we continue with two types of limits, one of them is set by the device and is constant throughout the session (cockpit calls them hard), and another that can be changed via the GUI (cockpit calls them soft). So closing this as fixed.

@iandobbie, I don't have anything against removing the currently named soft limits completely. I also never used them (back when I actually did wet lab). If you would like that, could you open a separate issue for it (and maybe invite other people to comment)?

closed 2020-06-02 by carandraug