microscope-cockpit/cockpit #526: Experiments require hardware trigger source

mickp, 2019-07-22:
It should be possible to have a software fallback that uses functions like "trigger_now" on the cameras to execute experiments.

mickp, 2019-11-14:
We should be able to support single-slice timelapse without a trigger source, as the action table should just be a bunch of camera and light toggles.

* Give a camera that supports soft_trigger an executor handler, which reports that it can run 1 line if soft_trigger is enabled.
* Make trigger source optional for light sources (may already be the case).

Having an executor handler on a camera should prevent creation of the dummy executor. 

There might be issues with there being light toggles in the action table, but nothing to run them ... maybe we could handle this by setting a light's trigger source to None, so the light handler also reports it can execute the toggle but it actually does nothing.

dstoychev, 2021-05-13:
This seems like a suitable place to request entirely software-based experiments, without any executor devices. Many modern cameras can control a light source directly, i.e. independently of Cockpit. For this reason, I think there should be an experiment where you can just soft trigger the camera and essentially do a time lapse. I believe this is more or less what Mick is requesting above, but without any light source involvement, just a sequence of images collected by soft triggering one or more cameras.

carandraug, 2021-05-13:
> This seems like a suitable place to request entirely software-based experiments, without any executor devices.  Many modern cameras can control a light source directly, i.e. independently of Cockpit. [...]

Do you mean having the camera as master device by setting a line high when it's exposing which the light sources are connected to?  The camera then is the executor device (has an executor handler). I guess this would require we settle on a trigger source abc in microscope which some cameras could have.

Note you'll still need to have cockpit set the light source to wait for such digital input.

dstoychev, 2021-05-13:
>Do you mean having the camera as master device by setting a line high when it's exposing which the light sources are connected to?

Yes.

But it doesn't make sense to me that camera would be an executor device. It cannot execute tables, set digital/analogue lines, etc. And there is no need to. It literally cannot be an executor device and there is no need to. The light sources should not be controlled by microscope/Cockpit, there is just no need for it and it complicates things too much. My expectation is to have a simple experiment which I can leave for, say, 60 minutes to take an image every second and collect them into a TIFF. No need for hardware triggers of any kind.

iandobbie, 2021-05-13:
Yo-yo might still want some light control from cockpit to change intensity or activate specific led wavelengths etc.

Ian