microscope-cockpit/cockpit #527: Deploy for Kseniya

mickp, 2019-07-22:
Kseniya is interested in deploying microscope and cockpit. This comment will contain the specification - I'll edit it as I obtain further details.

  camera:  Zyla 5.5
  stage:  ?
  light source: ?

mickp, 2019-07-31:
Windows installation tool created at mickp/cockpit-windows-installation
Waiting for feedback from Kseniya.

mickp, 2019-08-14:
Kseniya tried the install script. There was an issue with the URLs I'd used being unsuitable for anonymous cloning. I've fixed those, but now she has a problem with the script terminating almost immediately. I suspect it's a access control issue - the script needs to be run as Administrator by an account with admin privileges - but she's running Windows 10, and I've only tested on Windows 7.

iandobbie, 2019-10-23:
We tested this on windows 10 twice with the DPhil students this week and it worked very well. Only issue I found was once the system is installed as admin you then need the use the admin account to get a develop source tree to work. Overall a major success so I am closing.

closed 2019-10-23 by iandobbie