microscope-cockpit/cockpit #528: CameraDevice.__init__ not called

juliomateoslangerak, 2019-07-23:
At start I get this error. When I follow this, indeed the baseTransform does not exist cause CameraDevice.__init__ is never called.

Error in subscribed func devices.microscopeCamera.onObjectiveChange().  Traceback (most recent call last):
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\events.py", line 98, in publish
    subscribeFunc(*args, **kwargs)
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\devices\microscopeCamera.py", line 142, in onObjectiveChange
    self.updateTransform(transform)
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\devices\camera.py", line 76, in updateTransform
    base = self.baseTransform
AttributeError: 'MicroscopeCamera' object has no attribute 'baseTransform'

mickp, 2019-07-23:
Have you merged both the latest cockpit *and* the latest microscope?

juliomateoslangerak, 2019-07-23:
I believe so.

mickp, 2019-07-23:
Right - sorry. Looks like this is cockpit-side. We changed some transform-related stuff on the remotes fairly recently, to fix some other missing attribs, so I though it might be related to that. I'll take a look.

mickp, 2019-07-23:
I can't reproduce it here at commit @53d5a351b890c73afc6af6b7b8461eb911cfaca1: the super calls look right:
```MicroscopeCamera``` calls ```MicroscopeBase``` calls ```CameraDevice``` calls ```Device```.
```self.baseTransform``` is set correctly.
```updateTransform``` works correctly.
```onObjectiveChange``` throws no exception.

What python version are you using? 
Do you have old .pyc files hanging around?

mickp, 2019-07-23:
Maybe there's a config issue? Perhaps Device.__init__ is failing, causing CameraDevice.__init__ to exit early, and somehow this all happens before we're set up to see it?

juliomateoslangerak, 2019-07-23:
How could I test this? I went through Device.__init__ and it all looks fine. It just gets ip, port and uri, right? 
I'm using 3.6
I deleted .pyc but I suppose that if I run the debugger of pycharm it should recreate them...

mickp, 2019-07-23:
Yeah, the pyc files will be recreated after deletion, and that's fine. The problem is when there's an old one hanging around, and somehow the interpreter decides that it's current and uses that instead of regenerating it from the current py file.

I used the debugger with a breakpoint in ```MicroscopeCamera.__init__``` at:
```super(self.__class__, self).__init__(name, cam_config)```
Keep stepping into the ```super``` calls to see the execution order, and once you're in ```Device.__init__```, execute line by line to see that everything works fine, and work your way back up the call stack.

juliomateoslangerak, 2019-07-23:
That's what I did and everything looks fine. I get the ip, port and uri but by the time I'm out of  `super(self.__class__, self).__init__(name, cam_config)` CameraDevice.__init__ has not been called. I also have breakpoints there.

I don't know if this is normal but on pycharm's left margin I can see that CameraDevice and CameraDevice.__init__ are indeed subclassed/overridden (so is MicroscopeBase and its __init__). However, when I'm on MicroscopeCamera, the parent classes are not marked on the margin as it is usually the case

juliomateoslangerak, 2019-07-23:
I'll verify tomorrow with a bit more time but replacing the import syntax solves the problem
'from . import camera'
to
'from cockpit.devices import camera'

probably some path config issue

mickp, 2019-07-23:
Can you send me a screenshot of the left margin? I don't quite understand what you mean.

mickp, 2019-07-23:
Weird. I'd expect an error and module-parsing time if it couldn't do the import correctly.

juliomateoslangerak, 2019-07-24:
I would also expect that. It seems that there is no error at import cause the modules are found correctly. But the MRO are resolved differently, probably because they are considered as different classes. I evaluated `MicroscopeCamera.__mro__` using `from . import camera` vs `from cockpit.devices import camera`.

Using `from . import camera`
0 = {type} <class 'devices.microscopeCamera.MicroscopeCamera'>
1 = {type} <class 'cockpit.devices.microscopeDevice.MicroscopeBase'>
2 = {type} <class 'cockpit.devices.device.Device'>
3 = {type} <class 'devices.camera.CameraDevice'>
4 = {type} <class 'devices.device.Device'>
5 = {type} <class 'object'>
Using `from cockpit.devices import camera`
0 = {type} <class 'devices.microscopeCamera.MicroscopeCamera'>
1 = {type} <class 'cockpit.devices.microscopeDevice.MicroscopeBase'>
2 = {type} <class 'cockpit.devices.camera.CameraDevice'>
3 = {type} <class 'cockpit.devices.device.Device'>
4 = {type} <class 'object'>

I guess that the second is what we want. Still it does not explain why the relative import is missing CameraDevice. 
Is there any reason to use a relative import? I personally find it disturbing not to be specific about what is being imported.

juliomateoslangerak, 2019-07-24:
these are the icons I mean
[image]
You have an equivalent with the up pointing arrow in the child classes/methods. However, those last do not apear either using absolute imports. I guess pycharm does not deal with multiple inheritance

juliomateoslangerak, 2019-07-24:
May I suggest these changes here? 6af99dee29b3031d6766bc9757b959f0271d9cff
I changed the import format to be more coherent with the other imports and the more recommended use of super() of python 3.

mickp, 2019-07-24:
Right - I see what you mean, now. I thought you might have been talking about the "structure" panel.

I think JetBrains calls those icons next to the line numbers "gutter annotations." I don't see one for the MicroscopeCamera class whether I do:
```from . import camera```
or 
```from cockpit.devices.camera import CameraDevice``` (with a corresponding change to the class declaration).

I think a fully qualified import path is preferable. The local imports were introduced in the python-2 to python-3 fixes. In python 2, we just did "import camera", which first looks in the same folder as the current module. Now that cockpit is a package, though, a fully qualified path makes more sense - I'll merge 6af99de.

mickp, 2019-07-24:
Fixed by 6af99de.

closed 2019-07-24 by mickp

carandraug, 2019-07-24:
I can't replicate your issue with imports (Python 3.7.3 Debian Buster). I have opened a Python console on cockpit and checked the MRO. This is the current situation:

```
>>> import cockpit.devices.microscopeCamera
>>> cockpit.devices.microscopeCamera.MicroscopeCamera('', {'uri' : 'PYRO:TestCamera@127.0.0.1:8000'}).__class__.__mro__
(<class 'cockpit.devices.microscopeCamera.MicroscopeCamera'>,
 <class 'cockpit.devices.microscopeDevice.MicroscopeBase'>,
 <class 'cockpit.devices.camera.CameraDevice'>,
 <class 'cockpit.devices.device.Device'>,
 <class 'object'>)
```

And changing the imports to use the fully qualified names does not change it:

```
>>> import cockpit.devices.microscopeCamera
>>> cockpit.devices.microscopeCamera.MicroscopeCamera('', {'uri' : 'PYRO:TestCamera@127.0.0.1:8000'}).__class__.__mro__
(<class 'cockpit.devices.microscopeCamera.MicroscopeCamera'>,
 <class 'cockpit.devices.microscopeDevice.MicroscopeBase'>,
 <class 'cockpit.devices.camera.CameraDevice'>,
 <class 'cockpit.devices.device.Device'>,
 <class 'object'>)
```

mickp, 2019-07-24:
David - I couldn't replicate the problem, either, but 6af99de seems to fix things for Julio, doesn't break anything for me, and the changes in it make sense, so I've merged them.