microscope-cockpit/cockpit #529: Cockpit fails silently if it can't find FTGL.dll under Windows

mickp, 2019-07-29:
If FTGL.dll is not on the path or in the current working directory, it fails without displaying an error message or making a log entry.

carandraug, 2019-07-29:
Running on command line, one gets this error:

```
[...]
RuntimeError: Unable to load ftgl libary

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "cockpit/__init__.py", line 231, in OnInit
    cockpit.gui.ExceptionBox(caption='Failed to initialise cockpit')
UnboundLocalError: local variable 'cockpit' referenced before assignment
OnInit returned false, exiting...
```

Which explains why the error message is not displayed in the GUI. What is happening is that the `import cockpit.x.y` inside `CockpitApp.OnInit` are messing up the `cockpit` symbol that comes from import the cockpit modules in the module scope.

This can be workarounded by re-import `cocpkit.gui` when handling the exception or importing all the other stuff inside `CockpitApp.OnInit` as a separate symbol.

carandraug, 2019-07-29:
Which explains this comment on `cockpit/__init__.py`:

```
            # If I don't import util here, then if initialization fails
            # due to one of the other imports, Python complains about
            # local variable 'util' being used before its assignment,
            # because of the "import util.user" line further down combined
            # with the fact that we want to use util.logger to log the
            # initialization failure. So instead we have a useless
            # import to ensure we can access the already-loaded util.logger.
            from cockpit import util
            from cockpit import depot
```

But even doing this is not actually solving the issue of writing to the log file because the exception handling is using `cockpit.util` and not `util.` I'm guessing this worked before we had the cockpit namespace.

carandraug, 2019-07-29:
See my `import-scope-issue529` branch for a proposed fix (or carandraug/cockpit@49f997187489d68860e8f1446c22905471f7d703). This moves some imports out of the `OnInit` method and imports the window modules so it does not conflict with the gobal `cockpit` symbol.

We could reimport all the cockpit modules we needed inside `OnInit` like it was done before.  However, if we use importlib for the windows, then the problems gets fixed. And the main reason why using importlib is nice in this case, is to reduce code duplication since they all did something like:

```
import x
status.Update(x)
updateNum +=1
x.makeWindow
```

carandraug, 2019-08-02:
I have spoke with @mickp about the issue and pushed the fix proposed above.

It may be worth checking if the imports done within `CokpitApp.OnInit` could be done globally though.

closed 2019-08-02 by carandraug