microscope-cockpit/cockpit #530: Clarity filter configs an camera wavelengths.

iandobbie, 2019-08-07:
The clarity filter cube defin itions need to be able to set the excitation and emission wavelengths on the cameras so we can add it to the image file metadata for saving

mickp, 2019-08-07:
As discussed, the Clarity device does this in the same was as other filter wheels, through the config entry. Example:

```
[clarity]
type: cockpit.devices.aurox.Clarity
uri: PYRO:Clarity@127.0.0.1:8009
cameras: east
filters:
  1, Cy5, 680
  2, mCherry, 600
  3, Alexa488, 530
```

iandobbie, 2019-08-16:
Although no excitation wavelengths. A minor issues, except in SIM experiments. I suggest we close this issue as "Ian didn't read the documentation"

mickp, 2019-08-16:
I thought the Clarity was only in the emission path. Isn't it?

iandobbie, 2019-08-16:
Mick <notifications@github.com> writes:

> I thought the Clarity was only in the emission path. Isn't it?

Nope, in both.

Ian

mickp, 2019-08-16:
Doh. Of course. So the filterwheel is a wheel of cubes. Currently, the only excitation filters cockpit is geared up to use are ND filters. 

I think excitation filters would just be a metadata issue, but we'd need to extend the filter specification beyond ```name, emission-wavelength``` or ```name, ND%``` to something like ```name, excitation-wavelength, emission-wavelength```.