microscope-cockpit/cockpit #531: config.py default values

juliomateoslangerak, 2019-08-13:
Config.py defines some defaults for stage positions. It is noted that there should be sensible defaults, but I think that is waiting for disaster to happen. Should those be removed?
> 
            ## TODO: come up with sensible defaults.  These are historical.
            'dishAltitude' : '7570',
            'slideAltitude' : '7370',
            'slideTouchdownAltitude' : '7900',

juliomateoslangerak, 2019-08-13:
BTW, where should I override this? I tried to put some values on my zPiezo config but it does not get them...

> [zPiezo]
type: devices.stage.SimplePiezo
analogSource: cRIO
analogLine: 0
offset: 0
gain: 142.15
min: 0
range: 220
dishAltitude: 110
slideAltitude: 100
slideTouchdownAltitude: 120

carandraug, 2019-08-13:
> [...] Should those be removed?

Some sort of default is needed. Or how the thing is used needs to be changed. Any ideias?

> BTW, where should I override this?

Those should be on the stage section of your cockpit configuration (not on the device configuration). See the documentation section about configuration https://www.micron.ox.ac.uk/software/cockpit/config.html

mickp, 2019-08-14:
If we have any defaults, there's a risk of disastrous effects with a given stage. I thought we might be able to push the defaults off to individual devices, but I don't think this helps: these parameters come about due to the physical arrangement of a stage with respect to other components, so device-specific defaults could also have disastrous results.

As the code stands, if we remove the defaults, we add additional parameters that a user must provide before they can use cockpit.

These parameters are used to:
 * render certain elements in the MacroStageZ display;
 * provide targets for single-click moves in the MacroStage window.

I think the safest solution would be to set default values to None, and wherever they're used and found to be None, abort the operation:
 * don't draw the MacroStage elements that depend on these parameters, and fail silently;
 * don't respond to clicks on the relevant buttons, and fail with a message saying that the parameters have not been set.

juliomateoslangerak, 2019-08-16:
I agree with that solution. 
I would add that slide and dish are very specific types of sample. Those names should be generalized to something like 'sampleAltitude'. Even better, those should be customizable through the config it self to whatever the user wants it to be, but that might be considered as a new feature.