microscope-cockpit/cockpit #532: Canvas view scaling error

juliomateoslangerak, 2019-08-13:
I get this error with a black Image:
```
C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\imageViewer\viewCanvas.py:188: RuntimeWarning: invalid value encountered in double_scalars
  return - (self.vmin - self.dmin) / (self.dptp * self.scale)
C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\imageViewer\viewCanvas.py:344: RuntimeWarning: invalid value encountered in double_scalars
  return -1 + 2 * (val - self.lbound) / (self.ubound - self.lbound)
```

mickp, 2019-08-14:
This is probably a divide-by-zero error which can happen when the first image is full of a single value.

Options:
 a) limit the the determinant;
 b) set it to 1 if it is exactly equal to zero.

(a) works well for ints (just limit it to 2^-(bit depth), but maybe not for floats.
(b) is equivalent to (a) if there's one bit of noise about a single value; for one or more bits of noise, it will scale the noise to fill the dynamic range. If the image is truly single-valued, we don't scale it, and it just appears as a single grey level (which it is).

I think I prefer (b)?

mickp, 2019-08-14:
Fixed in 000b95a56de86291afbb587231b4389d05692768.

closed 2019-08-14 by mickp