microscope-cockpit/cockpit #533: Pyro searches local microscope while deserializing error

juliomateoslangerak, 2019-08-16:
I don't know if this belongs in cockpit or microscope.
When I try to change the parameter pixel encoding on the camera settings (I have two options Mono32 and Mono16), I get an error:
```
Traceback (most recent call last):
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\device.py", line 340, in <lambda>
    applyButton.Bind(wx.EVT_BUTTON, lambda evt: self.device.updateSettings(self.current))
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\interfaces\imager.py", line 79, in wrapper
    result = func(*args, **kwargs)
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\devices\microscopeCamera.py", line 115, in updateSettings
    self.proxy.update_settings(settings)
  File "C:\Users\julio\Applications\WinPython\python-3.6.5.amd64\lib\site-packages\pyro4-4.73-py3.6.egg\Pyro4\core.py", line 185, in __call__
    return self.__send(self.__name, args, kwargs)
  File "C:\Users\julio\Applications\WinPython\python-3.6.5.amd64\lib\site-packages\pyro4-4.73-py3.6.egg\Pyro4\core.py", line 467, in _pyroInvoke
    data = serializer.deserializeData(msg.data, compressed=msg.flags & message.FLAGS_COMPRESSED)
  File "C:\Users\julio\Applications\WinPython\python-3.6.5.amd64\lib\site-packages\pyro4-4.73-py3.6.egg\Pyro4\util.py", line 170, in deserializeData
    return self.loads(data)
  File "C:\Users\julio\Applications\WinPython\python-3.6.5.amd64\lib\site-packages\pyro4-4.73-py3.6.egg\Pyro4\util.py", line 451, in loads
    return pickle.loads(data)
  File "c:\users\julio\pycharmprojects\microscope\microscope\cameras\SDK3.py", line 37, in <module>
    _stdcall_libraries['ATCORE'] = ctypes.WinDLL('atcore')
  File "C:\Users\julio\Applications\WinPython\python-3.6.5.amd64\lib\ctypes\__init__.py", line 348, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 126] The specified module could not be found
```
It seems that pickle calls the local installation of micrsocope->SDK3->dlls to be abble to deseriailize the response of pyro. Is this expected?
The response of pyro follosws here and it contains a contains a AT_ERR_INDEXNOTAVAILABLE error.

```
b'\x80\x02cmicroscope.cameras.SDK3\nCameraError\nq\x00X\x0f\x00\x00\x00AT_SetEnumIndexq\x01K\x07\x86q\x02Rq\x03}q\x04(X\x05\x00\x00\x00errNoq\x05K\x07X\x07\x00\x00\x00fcnNameq\x06h\x01X\x0e\x00\x00\x00_pyroTracebackq\x07]q\x08(X#\x00\x00\x00Traceback (most recent call last):\nq\tX\xd6\x00\x00\x00  File "C:\\Users\\omxt\\Applications\\WPy-3661\\python-3.6.6.amd64\\lib\\site-packages\\Pyro4\\core.py", line 1421, in handleRequest\n    data = method(*vargs, **kwargs)  # this is the actual method call to the Pyro object\nq\nX\x87\x00\x00\x00  File "C:\\Users\\omxt\\PycharmProjects\\microscope\\microscope\\devices.py", line 373, in wrapper\n    result = func(self, *args, **kwargs)\nq\x0bX\xa2\x00\x00\x00  File "C:\\Users\\omxt\\PycharmProjects\\microscope\\microscope\\devices.py", line 603, in update_settings\n    super(DataDevice, self).update_settings(settings, init)\nq\x0cX\x90\x00\x00\x00  File "C:\\Users\\omxt\\PycharmProjects\\microscope\\microscope\\devices.py", line 358, in update_settings\n    self.settings[key].set(incoming[key])\nq\rXo\x00\x00\x00  File "C:\\Users\\omxt\\PycharmProjects\\microscope\\microscope\\devices.py", line 142, in set\n    self._set(value)\nq\x0eX\x8e\x00\x00\x00  File "C:\\Users\\omxt\\PycharmProjects\\microscope\\microscope\\cameras\\andorsdk3.py", line 63, in wrapper\n    return func(self, *args, **kwargs)\nq\x0fX\xa1\x00\x00\x00  File "C:\\Users\\omxt\\PycharmProjects\\microscope\\microscope\\cameras\\SDK3Cam.py", line 92, in setIndex\n    SDK3.SetEnumIndex(self.handle, self.propertyName, val)\nq\x10X\x8a\x00\x00\x00  File "C:\\Users\\omxt\\PycharmProjects\\microscope\\microscope\\cameras\\SDK3.py", line 220, in __call__\n    raise CameraError(self.name, res)\nq\x11X]\x00\x00\x00microscope.cameras.SDK3.CameraError: when calling AT_SetEnumIndex - AT_ERR_INDEXNOTAVAILABLE\nq\x12eub.'
```

juliomateoslangerak, 2019-08-16:
I came into the because I was getting systematically 12 bit images. There is something odd about the behavior.
While the setting `simple_pre_amp_gain_control` was in `16-bit (low noise high well capacity)` I have the option to change `pixel encoding` between `Mono16` and `Mono32`. I mean both options are shown in the menu but things fail to change as described above and I cannot set to Mono16. Though that is expected because the way data is encoded, it shouldn't give the option or through an error.
Now I set the `simple_pre_amp_gain_control` to `12-bit (high well capacity)` or `12-bit (low noise)` and I get to choose between `Mono12`, `Mono12Packed`, `Mono16` and `Mono32`. Following the same pattern, now I cannot set it to `Mono32`.
But now the weird thing comes. I go back to `16-bit (low noise high well capacity)` and `pixel_encoding` goes blanc though I get the option in the menu to change it to `Mono16` or `Mono32` (it is not greyed out). Either option does through the same error described above so the only way I can get out is to cancel. When I reopen the settings window it is blanc again. Also, as magic, now my images come in 16 bit.
Sorry it is messy but I hope you can follow. To summarize there seems to be an issue with:
- the simple_pre_amp_gain_control real state on the camera and what cockpit shows
- the readonly state and enum options of pixel_encoding as a function of the simple_pre_amp_gain_control.

mickp, 2019-08-16:
The Pyro behaviour is expected: any types passed over a connection must have definitions available at each end of the connection.

juliomateoslangerak, 2019-08-16:
I see. I installed the dlls and now the error is specified.
However, the issue with the camera settings remains. I will rename the issue

juliomateoslangerak, 2021-06-02:
I just checked on this after merging both microscope and cockpit and the issue is still present.
Every time I start a Zyla I have to change simple_pre_amp_gain control to 12 bits back and forth applying in between to get 16 bit images.

juliomateoslangerak, 2026-11-09:
I don't know where this got fixed or learned to synch microscope and cockpit, but this error is gone anyway. 

Closing