microscope-cockpit/cockpit #534: Stage position trails

iandobbie, 2019-08-20:
Is it worth implementing stage position trails so we can display previous stage positions. Should be relatively easy, record stage position in a list and then pass this to either the macrostage or the mosaic to display and draw the trail the stage has previously followed.

juliomateoslangerak, 2019-08-20:
This was a nice feature in the OMX software, but much less useful when all of the images are stored in the mosaic view. I would restrict this feature to the macrostage window.

iandobbie, 2022-05-26:
Added to the mosaic in [838b72e](https://github.com/iandobbie/cockpit/commit/838b72e6db28d60a8b6a9985a4f2cd2385b018ff) along with right click menu entries to turn them on and off as well as clear existing trails.

closed 2022-05-26 by iandobbie