microscope-cockpit/cockpit #535: Error on rendering mosaic canvas

juliomateoslangerak, 2019-08-20:
When I launch a mosaic I get this error (many times) and no images in the mosaic window. This is something has been always working so it must be a recent change on the code or on my installation...
Any idea?
```
Traceback (most recent call last):
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\mosaic\canvas.py", line 310, in onIdle
    megaTile.prerenderTiles(newTiles, self)
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\mosaic\tile.py", line 362, in prerenderTiles
    tile.render(viewBox)
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\mosaic\tile.py", line 243, in render
    glEnd()
  File "C:\Users\julio\Applications\WinPython\python-3.6.5.amd64\lib\site-packages\OpenGL\latebind.py", line 61, in __call__
    return self.wrapperFunction( self.baseFunction, *args, **named )
  File "C:\Users\julio\Applications\WinPython\python-3.6.5.amd64\lib\site-packages\OpenGL\GL\exceptional.py", line 46, in glEnd
    return baseFunction( )
  File "C:\Users\julio\Applications\WinPython\python-3.6.5.amd64\lib\site-packages\OpenGL\error.py", line 232, in glCheckError
    baseOperation = baseOperation,
OpenGL.error.GLError: GLError(
	err = 1286,
	baseOperation = glEnd,
	cArguments = ()
)
```

mickp, 2019-08-20:
That error code indicates an invalid framebuffer operation. What platform,
and have you recently updated your graphics card drivers?

mickp, 2019-08-20:
We used to use a hard-coded texture size for megaTiles. I changed that a while back to query the hardware and use something more sensible. I saw problems on the Mac with the Intel chipset reporting a larger texture size than it could really handle, but put in a fix for that.

I think some implementations might be reporting the maximum texture size that they *could* handle if enough video memory were available, without considering how much *is* available. There doesn't appear to be a common way to query how much video memory is actually available. So we could try to allocate then check for memory errors, or just set a hard-coded upper limit on the texture size.

Have a look at ```cockpit/gui/mosaic/canvas.py``` near line 162.
```
        vendor = glGetString(GL_VENDOR)
        tsize = glGetInteger(GL_MAX_TEXTURE_SIZE)
        if vendor.startswith(b'Intel'):
            # If we use the full texture size, it seems it's too large
            # for manipulation in a framebuffer on Macs with Intel chipsets.
            # GL_MAX_FRAMEBUFFER_WIDTH and _HEIGHT are not available, so we
            # just use a quarter of the max texture size, which has been found
            # to work intests on 2017-ish MacbookPro.
            tsize //= 4
```
Try printing out ```vendor``` and ```tsize```, and perhaps dividing ```tsize``` by 2, 4, 8 ... and so on.

juliomateoslangerak, 2019-08-21:
I'm on windows 10. My GPU is some reasonably performant nVidia. It is very possible that the drivers have been updated but I'm not aware of it. Since I passed to windows 10 OS updates are just a nightmare failing every single time.
vendor is 'NVIDIA Corporation'
tsize is (int32) 32768

If I divide tsize by 2 or 4 things work but I get the following error:
```
Exception ignored in: <generator object MosaicWindow.mosaicStepper at 0x00000230F9969150>
Traceback (most recent call last):
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\mosaic\window.py", line 638, in mosaicStepper
    def mosaicStepper(self):
SystemError: error return without exception set
```

mickp, 2019-08-23:
Yeah - it's taking ages to allocate massive textures. I think we need an upper limit on tsize.

iandobbie, 2020-01-26:
I think this is fixed in pull #550 so should be closed

closed 2020-02-13 by juliomateoslangerak