microscope-cockpit/cockpit #537: DSP abort is now totally broken.

iandobbie, 2019-09-24:
Any abort on the real DSP crashes the process with a windows error

OSError: [WinError -529697949] Windows Error 0Xe06d7363

mickp, 2019-09-24:
I saw this before, but thought I'd put in a fix. It's a C++ exception, so in one of the underlying DLLs.
Right - I did fix one instance of it. Are you running the latest commit from the py3 branch: a7eae56f013bc9e14f82ae4af76a26588ac5d0f7 ?

mickp, 2019-09-24:
As I remember, pyC67.Abort throws an exception from C++ if it's called during collection. I thought we were able to catch and ignore those, but maybe not. It writes the command "ABORT" to the DSP, then calls CollectionCleanup. We have no idea what the ABORT command does, and the source code we have for CollectionCleanup includes the line ```//todo ???``` ... so maybe the binary we've been running for years had some more stuff done in there?

mickp, 2019-09-24:
I can probably wrap a debug version that dumps more info to stdout. Let's try this tomorrow.

iandobbie, 2021-03-19:
This no longer and issue with deepsim as we are on the red pitaya and the cryosim system this works fine. I am closing

closed 2021-03-19 by iandobbie