microscope-cockpit/cockpit #540: OpenGL errors on Nvidia GTX 1080

iandobbie, 2019-10-17:
The Aurox computer has an Nvidia GTX 1080 and the mosaic tile update generates an openGL error 1286, seems to be an invalid frame buffer error from google search. I was rushing home but I will get a proper error message tomorrow.

iandobbie, 2019-10-22:
Reducing the max tile size by 16 fixes this issue so the graphics card is reporting more texture memory than it can actually access. Maybe we should limit to some set value (16385x16385?)

mickp, 2019-10-22:
As we discussed, it can address the memory, but it can't create a frame
buffer of that size. We need a sensible upper limit on tile size. 16384
will probably do it.

iandobbie, 2019-10-23:
Ok, I will try to do something like min(returned size, 16834) and see how it works on the Apple and GTX1080 cards which caused problems.

mickp, 2019-12-06:
Fixed by #550 ... I didn't see #546 until after I merged.

closed 2019-12-06 by mickp