microscope-cockpit/cockpit #541: Problem running on ubuntu.

iandobbie, 2019-10-21:
See comments below, I think this is a wx issue, and not strictly a cockpit or python problem. 

Suggestions:
- Create a list of package dependencies (e.g. in a .yml file), so users can create a virtual environment with
the correct packages.
- Please check if it runs on ubuntu (see error).?

-----------------------------------------------------------------------------------
Sequence of events

- I am on an ubuntu 18.04 OS.  
- I cloned the MicronOxford/cockpit.git repo.
- I ran the setup.py file as suggested in the readme, but it ran into missing module errors of pathlib2 and
subsequently pkg-config. I manually installed these. 
- Then I was able to run 'cockpit' in the terminal; first I got a missing ftgl.so error, solved by 'sudo apt-get
install libftgl-dev'. (libftgl2 does not work).
- Next, cockpit partly boots, but hangs in initialization phase, prompting an wx-related error, which freezes
my computer and requires manual reboot. (see error message below)

-----------------------------------------------------------------------------------
Error message 

wx._core.wxAssertionError: C++ assertion "m_window" failed at
/tmp/build/80754af9/wxpython_1547931003892/work/ext/wxWidgets/src/gtk/dcclient.cpp(2038) in
DoGetSize(): GetSize() doesn't work without window

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File
"/home/thijs/.conda/envs/cockpit/lib/python3.7/site-packages/cockpit-2.9.0+dev-py3.7.egg/cockpit/__init__.py",
line 165, in OnInit
    module.makeWindow(frame)
  File
"/home/thijs/.conda/envs/cockpit/lib/python3.7/site-packages/cockpit-2.9.0+dev-py3.7.egg/cockpit/util/intensity.py",
line 529, in makeWindow
    window = IntensityProfilerFrame(parent)
  File
"/home/thijs/.conda/envs/cockpit/lib/python3.7/site-packages/cockpit-2.9.0+dev-py3.7.egg/cockpit/util/intensity.py",
line 337, in __init__
    self.plotCanvas = plot.PlotCanvas(self, wx.ID_ANY, pos=(-1,-1))
  File "/home/thijs/.conda/envs/cockpit/lib/python3.7/site-packages/wx/lib/plot/plotcanvas.py", line 172, in
__init__
    self.OnSize(None)  # sets the initial size based on client size
SystemError: PyEval_EvalFrameEx returned a result with an error set

-- 

Thijs van der Plas

iandobbie, 2019-10-21:
There seems to be a suggestion that we create a conda based package file.
--
Create a list of package dependencies (e.g. in a .yml file), so users can create a virtual environment with the correct packages.
--
Surely the pypi setup file contains the same information and can be used by a virtualenv to gather the correct packages?

carandraug, 2019-10-21:
> There seems to be a suggestion that we create a conda based package file.
>
> Create a list of package dependencies (e.g. in a .yml file), so users can create a virtual environment with the correct packages.

I don't think we should do this in cockpit. conda is a, yet another, package manager. If we do that here, then by the same logic we would add Debian, homebrew, arch linux, fedora, chocolatey, and whatever equivalents. But package managers have their own places to do this.

> Surely the pypi setup file contains the same information and can be used by a virtualenv to gather the correct packages?

I haven't created a conda package, but I have created packages for other distros so my guess is no. There will be a separate file, with a format meant for conda, that duplicates this. There may be tools to create one from the other but then they will need to be kept in sync. Also, we have dependencies (like the ftgl libraries) that are not python packages.

juliomateoslangerak, 2019-10-22:
I have been using a bit of conda lately (miniconda actually) and the main advantage that I see is that it would take care of ftgl. 

https://anaconda.org/search?q=Ftgl

Maybe I am wrong but creating one of these yml files looks pretty easy...

iandobbie, 2019-10-22:
Yes I did a bit more research. Conda might be able to use its skeleton function to generate the relevant bits but it might well miss some magic, so need hand editing and maintaining. 

I don't think we should maintain another package format, especially as conda is open source but totally dependant upon a company. 

However, one tipbit we gained was on linux we had to install the ftgl-develop package (I will find the exact name) to get the ftgl bits to work.

mickp, 2019-10-22:
I remember discussing the ftgl requirement with David a while back. I think
we concluded that, using setup.py, there's no clean/easy way to deal with
prerequisites from outside python that need to be handled completely
differently from platform to platform.

carandraug, 2019-10-22:
> However, one tipbit we gained was on linux we had to install the ftgl-develop package (I will find the exact name) to get the ftgl bits to work.

What is happening here is that Debian (and so Ubuntu) splits C library packages into two. One has the shared libraries meant as dependencies for other things distributed with their package manager, that is, the versioned `libftgl.so.x.y.z`'s. Then, the stuff needed by those building new stuff from source are on a separate package, the unversioned `libftgl.so` and the header files.

We are effectively building cockpit from source so we need the libftgl package meant for development. The same happens in Fedora (and so Red Hat and CentOS). In Debian, the two package needed are `libftgl2` and `libftgl-dev` while on Fedora is `ftgl` and `ftgl-devel`. Other package managers might have different names.

iandobbie, 2019-10-22:
> I have been using a bit of conda lately (miniconda actually) and the main advantage that I see is that it would take care of ftgl.
> 

So have you created a yml file or are you just using the standard setup.py?

iandobbie, 2020-01-26:
This discussion was a couple of months ago and didnt provoke a great desire in anyone to add new package manager formats. Shall we just close this as "we support pypi and no other package managers"? We could also add a note to the install instructions to say you may need the ftgl-dev packages, but this is distribution dependant?

juliomateoslangerak, 2020-01-27:
I forgot to report that I tried conda to install ftgl and it didn’t work. Cannot recall the details but basically I ended up installing ftgl through the OS package manager.

mickp, 2020-01-27:
That's to be expected: FTGL isn't part of python, so won't be installed via a python package manager.

juliomateoslangerak, 2020-01-27:
Right, but it is expected to be installed through conda and so it was. But for some reason it did not work for cockpit...

carandraug, 2020-01-29:
> Right, but it is expected to be installed through conda and so it was. But for some reason it did not work for cockpit...

Where does conda place the ftgl.so file? Is this on Windows, MacOS, or Linux (which distro)? What do the following return on an interactive session?

```
import ctypes
ctypes.CDLL('libftgl.so') # WinDLL('ftgl.') if on Windows

import ctypes.util
ctypes.util.find_library('ftgl')
```

michalstepniewski, 2020-02-16:
What abt the freezing issue? Is it Still happening?

iandobbie, 2020-02-27:
The issue is that ftgl is not properly being loaded, or possibly the wrong version is being installed. Unfortunately it looks like anaconda isn't quite doing the right thing with the ftgl library. We have discussed this at length and decided that adding conda to the supported install methods is significant work for very little benefit. We support install via pip and setuptools but this does require explicit install of ftgl outside the python package installs.

carandraug, 2020-02-27:
> What abt the freezing issue? Is it Still happening?

I think this only happens with conda. Has anyone managed to reproduce the issue? I'm leaning on closing this issue as being a conda issue.

closed 2020-03-10 by carandraug