microscope-cockpit/cockpit #545: Why does the camera views always need a blank panel?

iandobbie, 2019-10-23:
The camera views have always had a blank panel as its menu bar was historically used to activate cameras. Now this is almost exclusively done through the main control window can we shrink the camera views to only display active cameras, not active cameras +1?

juliomateoslangerak, 2019-10-23:
I agree that we could save screen space there. However, I'm missing something that was accessed through a menu on the camera 'title', that was a quick menu to change ROI. Is this option gone?

iandobbie, 2020-04-02:
This is a trivial fix that I have tested and works. Can we please pull it and start reducing rather than increasing the number of issues.

carandraug, 2020-04-02:
I just looked at your fix Ian and it does work but is there any reason why inactive views (beyond the first for the blank view) are even added to the sizer in the first place? I changed it around a bit to avoid it, and seems to work and I think it also simplifies the logic. Take a look at my "wip-no-extra-blank-viewpanel" branch (or [1783b450](https://github.com/carandraug/cockpit/commit/1783b450a65020a0cfec7229f5c4e4f6f40c2000))

carandraug, 2020-04-03:
@iandobbie I pushed you fix together with a change to the comments that said that there was always an extra panel. I then pushed a second commit to not include hidden views in the sizer at all.

closed 2020-04-03 by carandraug