microscope-cockpit/cockpit #548: touchscreen view no longer updates when displayed.

iandobbie, 2019-12-06:
Something in the latest improvements to the mosaic view have killed any update to the touchscreen view. It only updates when something else, like a click in the window, causes a redraw. I guess it needs similar logic to the mosaic view with updates in idle periods, with an extra switch to not bother if it isn't displayed.

mickp, 2019-12-06:
I think this might be specific to MacOS - it works correctly under GTK. Check with latest version?

mickp, 2019-12-06:
Yep - still an issue on a Macbook Pro as of 9bdee0b5557e2d20f7e4549e1f25abb22ae4d27a

mickp, 2019-12-06:
The subscribed methods are getting called, and they are calling Refresh(), but glCanvases are not being redrawn. Non-gl stuff is redrawn correctly.
Refresh() should act recursively, calling Refresh() on all child objects in a window. This is working correctly for the MacroStage and Mosaic windows. In the Touchscreen window, Refresh is not being propagated to glCanvases. Perhaps they don't have invalidated regions?

PR #549 (2019-12-06): Fixes 548 with explicit call to glCanvas.Refresh()
Fixes #548. I added explicit calls to Refresh on the glCanvases if running on a Mac.
closed 2019-12-06 by mickp