microscope-cockpit/cockpit #552: Aurox issue on start-up

NickHallONBI, 2019-12-10:
The makeUI function asks for the settings to be described and asks for 'mode' but this doesn't seem to be a key in what I assume is a dictionary of settings.  

> Traceback (most recent call last):
  File "c:\cockpit\cockpit\__init__.py", line 142, in OnInit
    frame = mainWindow.makeWindow()
  File "c:\cockpit\cockpit\gui\mainWindow.py", line 386, in makeWindow
    window = MainWindow()
  File "c:\cockpit\cockpit\gui\mainWindow.py", line 203, in __init__
    item = thing.makeUI(topPanel)
  File "c:\cockpit\cockpit\devices\aurox.py", line 142, in makeUI
    mode_selector.Set(self.describe_setting('mode')['values'])
  File "C:\Program Files\Python37\lib\site-packages\pyro4-4.76-py3.7.egg\Pyro4\core.py", line 185, in __call__
    return self.__send(self.__name, args, kwargs)
  File "C:\Program Files\Python37\lib\site-packages\pyro4-4.76-py3.7.egg\Pyro4\core.py", line 476, in _pyroInvoke
    raise data  # if you see this in your traceback, you should probably inspect the remote traceback as well
KeyError: 'mode'

mickp, 2019-12-10:
Are you running the correct versions of both cockpit and microscope?
You need the latest master for cockpit, and MicronOxford/microscope#124 for microscope.

NickHallONBI, 2019-12-10:
Apologies, I had not fulfilled all the necessary start-up requirements. All the necessary steps that you emailed over have been completed. Cockpit will now start. However, we're still having problems with starting up the Aurox. 

If the config file is initialise with the Aurox device defined as;

> device(aurox.Clarity, 'localhost' , 7001, {'camera': testdevices.TestCamera,})

As per the email but with address changed, then when you try to switch to the 'calibration' mode it switches back to raw and fails with the following traceback:

> 2019-12-10 15:18:20,719:Clarity (microscope.devices):ERROR:PID 20544: in set_setting(mode):
Traceback (most recent call last):
  File "c:\microscope\microscope\devices.py", line 329, in set_setting
    self.settings[name].set(value)
  File "c:\microscope\microscope\devices.py", line 145, in set
    self._set(value)
  File "c:\microscope\microscope\filterwheels\aurox.py", line 176, in set_mode
    raise Exception ("Processing not available")
Exception: Processing not available

Additionally, the camera then refuses to fire and only reads noise, presumably because cockpit is no longer being provided with an URI for the camera. To try and rectify this, I changes the Aurox config to:

> device(aurox.Clarity, 'localhost' , 7001, {'camera': andorsdk3.AndorSDK3,})

However, this still produces the Aurox error and, additionally, results in an error when the camera is enabled which continuously occurs and can only be stopped by killing the device sever. Traceback provided below:

> ctypes.ArgumentError: argument 1: <class 'TypeError'>: wrong type
2019-12-10 15:24:49,848:Clarity (microscope.devices):ERROR:PID 20544: in _fetch_loop:
Traceback (most recent call last):
  File "c:\microscope\microscope\devices.py", line 554, in _fetch_loop
    data = self._fetch_data()
  File "c:\microscope\microscope\cameras\andorsdk3.py", line 376, in _fetch_data
    ptr, length = SDK3.WaitBuffer(self.handle, timeout)
  File "c:\microscope\microscope\cameras\SDK3.py", line 210, in __call__
    res = self.f(*ars)

mickp, 2019-12-10:
Not sure what's going on with the camera. The noise is probably when you had the Clarity configured to set up a TestCamera instance. That type error in SDK3 is weird. I'm not sure, but it looks like something is being passed to (or back from) a DLL function with the wrong type. Is this a USB or CameraLink Zyla?