microscope-cockpit/cockpit #553: centroid tool / beamdelta in cockpit

carandraug, 2019-12-11:
We have recurrent discussion about getting the centroid alignment tool available straight from cockpit.

On github alone there is PR #495 which first introduced the idea (folded into the changes for AO tools) and was closed without being merged. Then there is PR #502 which picks the centroids logic from the previous PR and which we are not keen on merging and was closed (see review on PR), and PR #521 which is slightly different from PR #502 but it's not clear why  and it doesn't address the previously raised PR anyway.

We now have the whole logic into BeamDelta which is a separate program. Still, it seems there's want to be able to start an alignment procedure from within cockpit.

carandraug, 2019-12-11:
In the ideal situation we would have some sort of plugin system for cockpit but we have no plans or designs for such thing yet. Mick was mentioning an idea this morning that would implement centroid alignment from cockpit without adding extra complexity to the `viewCanvas`.

As an alternative, I'm thinking we could just start the BeamDelta program from cockpit. A quick proof of concept on my [`wip-beamdelta` branch](https://github.com/carandraug/cockpit/tree/wip-beamdelta) which certainly would need some massage.

iandobbie, 2023-07-21:
This is 3 1/2 years old and nobody moved it forward. BeamDelta is a pretty reliable stand alone tool and Davids branch above basically just uses depot to find the camera URI's and starts beamdelta with that. Maybe a better solution is to add the ability to read camera configs from a cockpit depot file to beamdelta.

closed 2023-07-21 by carandraug