microscope-cockpit/cockpit #555: experiment cleanup reset triggers of camera that arent used

iandobbie, 2020-01-20:
The experiment cleanup resets all camera triggers which means that a crashed or not connected camera will crash the cleanup and prevent image reordering. we should just reset the triggers on the cameras we are using for the experiment.

iandobbie, 2020-01-26:
The problem code is 

    def cleanupAfterExperiment(self):
        """Restore settings as they were prior to experiment."""
        if self.enabled:
            self.updateSettings(self.cached_settings)
            #self._proxy.update_settings(self.settings)
            self._proxy.enable()
         self.handlers[0].exposureMode = self._proxy.get_trigger_type()

The final function call setting the exposure mode fails if the remote is not present or crashed. Why do we need to do this is the camera was not used in the experiment and not enabled. Can we just move the final line into the if self.enabled: block?

carandraug, 2020-01-29:
The problem code that ian is mentioned is a method of the `microscope.devices.microscopeCamera.MicroscopeCamera` class.

carandraug, 2020-02-19:
I took a look at this now. The `cleanupAfterExperiment` should only be being called on cameras being used on the experiment. Here's what I found:

1. `cockpit.gui.dialogs.experiment.experimentConfigPanel.ExperimentConfigPanel.runExperiment` gets a list of all enabled cameras.
2. the list of enabled cameras is used to construct `cockpit.experiment.structuredIllumination.SIExperiment` which the parent class `cockpit.experiment.experiment.Experiment` places under `self.allHandlers`.
3. so this `allHandlers` is not all device handlers, it's all the handlers being used for the experiment

Checking if the camera is disabled at the end of the experiment should not be a fixed because that check already happened before.

Would be nice to get a set of instructions that could reliably replicate the issue so that we can get a proper fix.

Still, there is the issue that a crash during one of the devices cleanup prevents the cleanup of the other devices or any other cleanup that might be important to put the system in a safe state.

iandobbie, 2020-04-07:
I understand David's point that this shouldn't be happening but it definitely did, reproducibly kill the experiment cleanup, when the Ximea camera was not active in the experiment. The suggested fix did solve the problem and prevented Nick having any more un-reordered files. 

I could reliably reproduce it by starting cockpit. unplugging the ximae camera and then running a SIM experiment, not using the camera.

Unfortunately with limited hardware it is not easy to test. I will try to setup a ximea and a red pitaya trigger source after easter to test these types of things.