microscope-cockpit/cockpit #556: SLM configurations.

iandobbie, 2020-02-04:
The diffraction angle always resets to some default value (0.35). We ned to be abel to hgave a configuration which is applied every time we start on a given system.

juliomateoslangerak, 2020-03-31:
Shouldn't this be changed with the objective? That is, should be a config setting of the objective.
Also the settings to achieve relative intensities on the different orders should be configurable, maybe per wavelength...

iandobbie, 2020-03-31:
Yes, both diffraction angle and a bunch of other stuff should be configurable. This really requires the slmserver to be moved into microscope. I think this should probably be prioritised.

Are you using a different SLM? If so this would be a major advantage as we could then have a more general interface for SLMs. 

The microscope deviceserver is actually based on the slmserver so it should be quite easy to convert the slmserver to a more standard deviceserver module.

juliomateoslangerak, 2020-03-31:
I'm actually using BNL's OverDrive Plus. It's very fast but, while the drivers / sdk is named similarly, it works different. I had to rewrite part of the interface. The main differences are, If I recall properly, 
- the way references are taken
- there is a GPU based precalculation of the transient images
- works with a 8bit controller so images are 8 bit
All of that can be fit into the current interface, but behind the scenes it's different. Here is what I run in the branch BNL_ODP
https://github.com/juliomateoslangerak/bnsdevice.git

juliomateoslangerak, 2023-12-22:
Hi @iandobbie and @carandraug, 
I'm looking into the cockpit side of the SLM control cause I want to implement a quick&dirty implementation of RIM. But looking forward into moving the SLM into microscope and extensibility, some thoughts.

Currently we are adding through ```table.addAction(curTime, self.slmHandler, (angle, phase, longestWavelength))``` and that turns into a SLM-remote call to ```self.asproxy.set_sim_sequence(sequence)```.

My feeling is that the only flexible enough solution is that is should be the experiment module the one generating the patterns and send them into the SLM along with the wavelength. So ```table.addAction(curTime, self.slmHandler, (phase_image, wavelength))``` ->  ```self.asproxy.set_sequence(sequence)```

We discussed about this in some meeting and I recall that there was a concern about the time it would take to send the patterns. My take on this is that, if this delay is at the experiment start, that would be acceptable. We would have to do something with the multi-site experiment where a new experiment instance is generated every time. But that is another issue.

Would you agree with this approach?

iandobbie, 2024-01-03:
I think transfer time for the planned patterns is relatively minor issue, and we did implement an image upload feature for the ISO-Sense paper, way back.