microscope-cockpit/cockpit #557: Imports in cockpit/__init__ are not well referenced

juliomateoslangerak, 2020-02-13:
My app fails to start cause some of the imports in the cockpit/__init__ file are not called properly.
In particular events and depot. 

BTW, I would suggest a more concise import of the cockpit modules. something like this, leaving cockpit.config separated cause it might be prone to confusion.
``` python
import cockpit.config
from cockpit import events, depot, gui
from cockpit.interfaces import imager, stageMover
from cockpit.util import files, logger, userConfig
```

carandraug, 2020-02-13:
I don't understand what you mean with them not being called properly. They are imported like:

    ...
    import cockpit.depot
    import cockpit.events
    ...

Can you provide an example that shows the issue?

juliomateoslangerak, 2020-02-13:
Yes, but later on (lines 101-105) events and depot are not called correctly.
``` python
import cockpit.depot
import cockpit.events
...
events.subscribe('program startup failure', self.onStartupFail)
events.subscribe('program exit', self.onExit)

depot_config = self.Config.depot_config
depot.initialize(depot_config)
```
I think it looks cleaner to do 
``` python
from cockpit import depot, events
```

carandraug, 2020-02-13:
I see now. The calling is incorrect but I don't know why it works here and not there. Fixed it by calling it properly, i.e. `cockpit.x` in 700ffde1c60603d.

I disagree with the `from cockpit import x, y` because some of the modules have very general names like `files` and `imager` which are common names and may cause confusion in the future.

closed 2020-02-13 by carandraug