microscope-cockpit/cockpit #558: Intensity profile tool not python 3 compatible

juliomateoslangerak, 2020-02-14:
The intensity utility is not python 3 compatible and does not run as a stand alone application.

I have a tested fix here 9de2d0492f93b979c085591a3ce3d3970a743090. Mainly the Mrc import and some floats that are passed as indexing values.

iandobbie, 2020-02-17:
Turns out Nick has fixed this on DeepSIM but se think the updates got buried in an AO update that hasnt been pulled yet. I have asked him to compare versions and see if there are any missmatched fixes and harmonies them. 

Other than that this should be pulled as it is stupid having a python 2 helper function in a python3 project!

carandraug, 2020-04-03:
Thank you for the fix Julio. I rebased your commit (now 871402cfd885372d32e2 ) and pushed it. Closing this as fixed.

@iandobbie wrote:
> Turns out Nick has fixed this on DeepSIM but se think the updates got buried in an AO update that hasnt been pulled yet.

I'm not sure what's the status of @NickHallONBI 's fixes for AO, but I couldn't find anything about this on his PR #518 .

closed 2020-04-03 by carandraug