microscope-cockpit/cockpit #559: SIM experiment reordering failure

NickHallONBI, 2020-02-18:
Cockpit sometimes fails to reorder SIM experiment data in the order that SoftWoRx expects. It is unclear why this happens. It has happened before in the past with short exposures (10ms) but I was collecting data at 50ms exposure on both the cameras and it happened again. 

I happened to be looking at the file in the file explorer and noticed that, whereas normal experiments have the data file of size 0 KB until the experiment is finished and then the file becomes whatever size it's supposed to be, for this experiment (and I'm assuming the other instances where this issue occurs) it saves ~100 MB of ~5 GB file part way through the experiment. I'm guessing that since the file isn't size 0 any more, when the data tries to write the reordered data to the file it fails and hence the data is never reordered. How or why this error doesn't stop the data being written period I don't know.

Here is the traceback:

> Error in subscribed func cockpit.devices.microscopeCamera.cleanupAfterExperiment().  Traceback (most recent call last):
  File "c:\cockpit\nick\cockpit\cockpit\events.py", line 98, in publish
    subscribeFunc(*args, **kwargs)
  File "c:\cockpit\nick\cockpit\cockpit\devices\microscopeCamera.py", line 131, in cleanupAfterExperiment
    self.handler.exposureMode = self.proxy.get_trigger_type()
  File "c:\python37\lib\site-packages\pyro4-4.75-py3.7.egg\Pyro4\core.py", line 185, in __call__
    return self.__send(self.__name, args, kwargs)
  File "c:\python37\lib\site-packages\pyro4-4.75-py3.7.egg\Pyro4\core.py", line 476, in _pyroInvoke
    raise data  # if you see this in your traceback, you should probably inspect the remote traceback as well
ximea.xiapi.Xi_error: ERROR 1: Invalid handle
Exception in thread Experiment-cleanup:
Traceback (most recent call last):
  File "C:\Python37\lib\threading.py", line 917, in _bootstrap_inner
    self.run()
  File "C:\Python37\lib\threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "c:\cockpit\nick\cockpit\cockpit\experiment\structuredIllumination.py", line 497, in cleanup
    self.reorder_img_file()
  File "c:\cockpit\nick\cockpit\cockpit\experiment\structuredIllumination.py", line 414, in reorder_img_file
    doc = cockpit.util.datadoc.DataDoc(self.savePath)
  File "c:\cockpit\nick\cockpit\cockpit\util\datadoc.py", line 83, in __init__
    self.image = Mrc.bindFile(filename)
  File "c:\cockpit\nick\cockpit\cockpit\util\Mrc.py", line 73, in bindFile
    a = Mrc(fn, mode)
  File "c:\cockpit\nick\cockpit\cockpit\util\Mrc.py", line 109, in __init__
    self.doDataMap()
  File "c:\cockpit\nick\cockpit\cockpit\util\Mrc.py", line 176, in doDataMap
    adjusted_shape = adjusted_data_shape(self.data.size, header_shape)
  File "c:\cockpit\nick\cockpit\cockpit\util\Mrc.py", line 1077, in adjusted_data_shape
    % (numel, str(shape))))
ValueError: data too large (2524446720 elements) for proposed shape (4815, 2, 512, 512)

NickHallONBI, 2020-02-18:
Unsure whether to delete the traceback, but I just collected another set of data with 100 ms exposure and Cockpit still failed to reorder it. However, this time it failed silently (no error message or warning). I only discovered on opening the file,

juliomateoslangerak, 2020-02-18:
I have exactly the same issue. To me it also seems to be correlated with the acquisition speed. And I do not get any error reported. 
My best guess was that it is the network that is clogged. Also because upon fast acquisitions sometimes I loose the socket to my FPGA. 

Is there something in place that guarantees the order of arrival of the images through the network?

What network speed do you have? I have a 1G but, as soon as I receive some cables that are held back in China, I’m moving to 10G from cameras to switch and 20G from switch to cockpit PC. 
I hope this helps and indicates if the source of the problem is network performance but not solve the issue.

iandobbie, 2020-02-19:
We have no guarantee of image order. We have discussed it but not implemented this. Suggestions of how to do this are welcome! 

Any dropped frames will lead to a reorder failure as the experiment end is never called and shouldn't be as we don't know which frame was dropped. However Nick's issue is different as all the frames are present so I think the finalise experiment code is crashing elsewhere and the previous occurrences have been due to the trigger reset issue in #555. I will generate a pull request and merge this change and see if it helps.

juliomateoslangerak, 2020-02-19:
I think my issue is the same as Nick's. I also have all the frames but their order are scrambled. In chunks actually. there might be some 6 images in one angle and then another few in another...
Indeed when some frames are not received I get an error reporting an unexpected amount of data for an expected data shape. I get those in other circumstances.

I had a quick look to the code and saw that there is one thread saving the file and a second thread for cleaning and only reordering the data after that. if it were really like that, and the saveThread is saving to disk, we should see the file saved on disk before it being reordered...

iandobbie, 2020-04-02:
I am sure the Nick's issue was the camera trigger resets for a crashed camera. I have pushed a fix (#555) that only resets triggers on used cameras. I know that David things the camera list should not include unused cameras but this is not the case on DeepSIM. 

Julio's issue may well be different and we should receive both image data and time stamps on the returned images so should able to order them "properly" however I don't think this would be trivial.

iandobbie, 2021-03-19:
This was an issue with experiments not properly running the cleanupAfterExperiment() code, so not getting to the reorder images code. I am closing as no longer relevant

closed 2021-03-19 by iandobbie