microscope-cockpit/cockpit #561: Unable to run experiment on XimeaCamera

dstoychev, 2020-03-10:
I am unable to run experiments using XimeaCamera. The problem seems to be that the trigger mode is not set, and then the getExposureMode function of the handler returns None, which in turn raises the error below.

Traceback (most recent call last):
  File "C:\micropit\cockpit\cockpit\gui\dialogs\experiment\singleSiteExperiment.py", line 130, in onStart
    if self.panel.runExperiment():
  File "C:\micropit\cockpit\cockpit\gui\dialogs\experiment\experimentConfigPanel.py", line 519, in runExperiment
    return self.runner.run()
  File "C:\micropit\cockpit\cockpit\experiment\experiment.py", line 218, in run
    self.createValidActionTable()
  File "C:\micropit\cockpit\cockpit\experiment\experiment.py", line 278, in createValidActionTable
    self.table = self.generateActions()
  File "C:\micropit\cockpit\cockpit\experiment\zStack.py", line 90, in generateActions
    curTime = self.expose(curTime, cameras, lightTimePairs, table)
  File "C:\micropit\cockpit\cockpit\experiment\experiment.py", line 613, in expose
    raise Exception ('%s has no trigger mode set.' % camera)
Exception: <camera named aurox-camera in group universal camera> has no trigger mode set.

dstoychev, 2020-03-11:
The problem seems to be that the [XimeaCamera](https://github.com/MicronOxford/microscope/blob/7730fdbd2b303a0b004badc60fb5e32f954d5af8/microscope/cameras/ximea.py#L146) device in microscope doesn't implement the [get_trigger_type()](https://github.com/MicronOxford/microscope/blob/7730fdbd2b303a0b004badc60fb5e32f954d5af8/microscope/devices.py#L824) interface. This becomes an issue in cockpit whenever a Ximea camera is enabled via [enableCamera()](https://github.com/MicronOxford/cockpit/blob/700ffde1c60603d7701d7384d0c443fc815bc857/cockpit/devices/microscopeCamera.py#L192), which overwrites the exposureMode property to None because the get_trigger_type() interface is not implemented.

The solution is to agree on a common interface between all camera devices in microscope for getting the current trigger type/mode. If the interface is renamed cockpit will also need to be changed.

carandraug, 2020-03-11:
As we spoke a bit earlier, cockpit makes use of the `get_trigger_type` which is not an abstract method so it is possible that a microscope won't implement it. This is the case for `XimeaCamera` which makes use of the `TriggerTypeMixIn` which has a `trigger_type` property. I suggest that cockpit tries `TriggerTypeMixIn.trigger_type` if that fails check if `CameraDevice.get_trigger_type()` returns something other than `None`.

On the microscope side, would be nice to do it in one way only but that's a discussion to have on the microscope repo.

dstoychev, 2020-03-19:
Do you mean ``TriggerTargetMixIn``? There doesn't seem to be a ``TriggerTypeMixIn``. All the camera classes, except for ``XimeaCamera``, seem to have the ``get_trigger_type()`` method defined. It is also defined for the underlying ``CameraDevice`` class, although not as an abstract method. As it stands now, ``XimeaCamera`` class has got the method inside of it, it is simply not implemented properly.

From cockpit's point of view, there should be a single interface for querying the trigger type. Not two interfaces that do the same thing. If you think this one interface should be the ``trigger_type`` attribute, then all the other camera classes should implement it as well. Piecemeal changes to the interface are breaking software changes and they should be avoided.

carandraug, 2020-03-20:
> Do you mean `TriggerTargetMixIn`?

Yes.

> [...] It is also defined for the underlying `CameraDevice` class, although not as an abstract method. As it stands now, `XimeaCamera` class has got the method inside of it, it is simply not implemented properly.

If it is not annotated as an abstract method, then it's not required and may not work. I believe the original idea behind that method was for the case of cameras that didn't support hardware trigger. The `CameraDevice` class has many of those.

> [...] Piecemeal changes to the interface are breaking software changes and they should be avoided.

The interface has not been broken. The method `CameraDevice.get_trigger_type` has not been reimplemented by the subclass and so returns `None` as it should.

I think I know what you mean though but making more global changes is difficult when we can't test on all the supported hardware. Anyway, I have previously proposed to just make all cameras and mirrors subclass from `TriggerTargetMixIn` so it works for all devices that may accept any type of trigger, even if software only. I will prepare something this weekend.

carandraug, 2020-03-23:
I have looked at making all the cameras in microscope use the `TriggerTargetMixIn` but only managed to get the andorsdk3 to do it. See my `wip-triggertargetmixin-to-all-cameras` branch. I have found that the cameras have even more trigger modes that are no supported.

On the other hand, I also made the current ximea use the interface that cockpit is expecting, see my `wip-cockpit-issue-561` branch.

iandobbie, 2020-04-02:
Whats the current status on this? I actually have a ximea camera and a red pitaya at home so I could do some testing if needed.

dstoychev, 2020-04-03:
@iandobbie I would say check @carandraug's branch from the message above yours and see if it solves the issue (it should do).

carandraug, 2020-08-10:
I pushed the proposed fix above that implements `XimeaCamera.get_trigger_type` which makes it work now. On the long run, we should support better trigger types and modes. Closing this as fixed.

closed 2020-08-10 by carandraug