microscope-cockpit/cockpit #563: Python 3.8 support

dstoychev, 2020-03-24:
Opening this to keep track of all changes necessary to support python 3.8+. For now, the only issue I have encountered is the way DLLs are loaded on Windows, which breaks FTGL:

https://docs.python.org/3.8/whatsnew/3.8.html#changes-in-the-python-api

>DLL dependencies for extension modules and DLLs loaded with ctypes on Windows are now resolved more securely. Only the system paths, the directory containing the DLL or PYD file, and directories added with add_dll_directory() are searched for load-time dependencies. Specifically, PATH and the current working directory are no longer used, and modifications to these will no longer have any effect on normal DLL resolution. If your application relies on these mechanisms, you should check for add_dll_directory() and if it exists, use it to add your DLLs directory while loading your library. Note that Windows 7 users will need to ensure that Windows Update KB2533623 has been installed (this is also verified by the installer). (Contributed by Steve Dower in bpo-36085.)

dstoychev, 2020-03-24:
#564

carandraug, 2020-03-25:
> Opening this to keep track of all changes necessary to support python 3.8+. 

I think it's better to have only one issue on each issue record. It will avoid having the discussion of multiple issues interleaved (which will make it harder to follow) and lets us can refer to a specific issue by number. Also, in this specific case what would we do if after closing this we find a new issue related to python 3.8?  Let's keep one issue per issue.

---

For context of that release note, there is the [associated issue on Python's bug tracker](https://bugs.python.org/issue36085é) and Microsoft's [LoadLibraryExW function docs](https://docs.microsoft.com/en-us/windows/win32/api/libloaderapi/nf-libloaderapi-loadlibraryexw) (from reading the Python source, the new default are the flags `LOAD_LIBRARY_SEARCH_DEFAULT_DIRS | LOAD_LIBRARY_SEARCH_DLL_LOAD_DIR`).

carandraug, 2020-03-25:
One of the main reasons for the change in Python is to not search DLLs in `%PATH%` because that's a security issue and deprecated by Microsoft (so the Python bug says, following their link to Microsoft documentation I only found that Microsoft recommends against it). Using `ctypes.util.find_library` as proposed on #564 we will use the full dll filepath when loading the  libary which is one of the recommendations. However, `ctypes.util.find_library` finds the filepath by searching `%PATH%` which defeats the whole change in Python.

Searching online, it seems that the recommended way to install a library on Windows is to put it on the system path or the application directory. Since ftgl is library only, it seems that it should go into the system path. Since the system path is still searched on windows to search for dll's can this be fixed on the ftgl installer?

It is well known that I'm not super familiar with Windows so my interpretations might be wrong.

dstoychev, 2020-03-25:
As far as I know, there is no installer on Windows for FTGL. You just build the library from source or grab a pre-built DLL or LIB file. I did the latter and placed the DLL in a directory that's in PATH. Python still couldn't find it. I tried to add the directory to PYTHONPATH too, unsuccessfully.

It seems that the problem is, as you said in your previous comment, that the default flags passed to the ``LoadLibraryExW`` function have changed and because of that the default Windows search paths are not used to find the DLL. I just tried to do
```
import ctypes
_ftgl = ctypes.WinDLL("ftgl", winmode=0)
```
and it works. Unfortunately, this ``winmode`` parameter is new to Python 3.8 so using it will break compatibility with older versions.

carandraug, 2020-03-25:
> As far as I know, there is no installer on Windows for FTGL. You just build the library from source or grab a pre-built DLL or LIB file. I did the latter and placed the DLL in a directory that's in PATH. Python still couldn't find it. I tried to add the directory to PYTHONPATH too, unsuccessfully.

What if you place the dll in `%windows%\system32` does that work? What I understood is that on windows, libraries meant for other programs should go there, and my reading of Python's code is that it should work.

> I just tried to do
> ```
> import ctypes
> _ftgl = ctypes.WinDLL("ftgl", winmode=0)
> ```
> and it works.

This is interesting because Python's documentation states that `winmode` default is zero. I have sent a [PR to cpython](https://github.com/python/cpython/pull/19167)

> Unfortunately, this winmode parameter is new to Python 3.8 so using it will break compatibility with older versions.

Agreed.

I'm not sure this is a bug of ours, maybe if we were distributing cockpit installers it would. Can you fix the distribution of ftgl?

carandraug, 2020-07-05:
Closing since ftgl is no longer required (see #615).

closed 2020-07-05 by carandraug