microscope-cockpit/cockpit #565: remove status light window

carandraug, 2020-03-30:
We have spoken multiple times about reducing the number of windows in cockpit. One of the first to be mentioned are the status light window which maybe could be merged into some other window and it has been proposed to implement them as status bar on the main window.

carandraug, 2020-03-30:
I have an initial implementation of this in my `wip-statusbar` branch (or see https://github.com/carandraug/cockpit/commit/d2705187666893cba71e620a9d673009b7fdd3d5 ) I made the font of the status bar larger than the default because I thought the default was too small.

One problem with using the status bar is that I could not find a method to set the colour of individual panes, only the background colour of the whole bar. The documentation suggests it might be possible to create our own status bar class with their own status bar pane, but I'm not sure. Once we know better what we want, we can take a look at it.

iandobbie, 2020-03-31:
Is it possible to have the same status bar on the touchscreen window as well? Maybe we need a wrapper around the updated functions to update both is the touchscreen window is visible.

carandraug, 2020-03-31:
> Is it possible to have the same status bar on the touchscreen window as well?

Should be no problem. Just do:

```
self.SetStatusBar(cockpit.gui.mainWIndow.StatusLights(parent=self)) 
```

on the touchscreen frame.

carandraug, 2020-03-31:
Oh! And before I forget:

carandraug wrote:
> One problem with using the status bar is that I could not find a method to set the colour of individual panes, only the background colour of the whole bar. The documentation suggests it might be possible to create our own status bar class with their own status bar pane, but I'm not sure.

I spoke with wxWidgets devs and they told me it is not possible to have a status bar where each field/pane have different background colours. However, nothing stops us from simply creating a wxPanel that does wnat we want and add it to the bottom of the frame.

I have also spoke with wxPython devs who mentioned that there used to be an [`EnhancedStatusBar`](https://wiki.wxpython.org/EnhancedStatusBar) that supported that but those are wxPython only, last update of it was in 2005, and my previous experience with wxPython specific features is that they are not well maintained. If we go down the route of enforcing background colour of the status light, then I think it's better to have our own as suggested above.

iandobbie, 2020-03-31:
I added the relevant code to the touchscreen but your fork is isolated so I can't send you a pull request. 

I added at line 412 of gui/touchscreen.py

        #add status bar to bottom of touchscreen display
        self.SetStatusBar(cockpit.gui.mainWindow._StatusLights(parent=self))

carandraug, 2020-03-31:
> I added the relevant code to the touchscreen but your fork is isolated so I can't send you a pull request.

Does that mean it worked? I don't know how it looks in Windows and MacOS.  Are we moving forward with this approach of replacing status light window to a status bar then? If so, do we give up on the colours or do we implement our own status bar look-alike panel with colours?

Also, at the moment, all of the status light are during experiments ('device waiting' and 'image count'). When I was working on the experiment panel (that change got too complex and modified how experiments were setup and so is postponed since it's not essential), one proposal was to keep the experiment window open until the experiment ends. With that change, instead of a status lights window, the messages appeared on the experiment progress bar.  When that gets finished, then maybe the colour is no longer important (I'm assuming the colour is important to raise attention to them) because being in the progress bar itself is enough to raise attention to them.

carandraug, 2020-03-31:
> at the moment, all of the status light are during experiments ('device waiting' and 'image count'). 

I was wrong. The nanomover also publish its position after a move. Still, we can have a normal status bar for simpler updates, experiment window for experiment updates, and dialog windows for really important messages that the user should not miss (I don't think there's any like that at moment).

You can find all the current status light messages with `grep -r UPDATE_STATUS_LIGHT cockpit`

iandobbie, 2020-04-01:
The nanomover publishes updates but these don't go to the status bar, but to the gui objects so they know to do gui updates.

iandobbie, 2020-04-01:
And yes the status bar did appear to work but as I can't run experiments I couldnt really confirm it was running properly. I think it is a good piece of progress towards window reduction see #567. We should discus this in the design discussion on 2/4 .

carandraug, 2020-04-01:
> The nanomover publishes updates but these don't go to the status bar, but to the gui objects so they know to do gui updates.

They do send updates to the status lights. I have spoken with @iandobbie on the lab meeting today and this is not meant to happen so I removed it with 5ed6647ed7ad. With this change, the only status lights are 'device waiting' and 'image count', all related to experiments. There is also a 'room light' on `cockpit/devices/ni6036e.py` but it's commented out.

carandraug, 2020-04-01:
> I can't run experiments I couldnt really confirm it was running properly.

As spoken earlier today, it is possible to have a fake DSP device and run fake experiments. @iandobbie, you mentioned a repository for sample config files but I can't find it.  There's is the [configs repo](https://github.com/MicronOxford/configs/) which is private and out of date, and there's the sample-configs directory in the cockpit repo which is for individual devices and also out of date.

iandobbie, 2020-04-01:
I started to do this but obviously didn't actually get it to micron. I remember asking Nick to put his configs into it and I also planned to import the Diamond CryoSIM setups.

H ave the start of this at https://github.com/iandobbie/microscope-cockpit-configs but only has DeepSIM depot.conf, no microscope configs, or other systems. We need to store these, where is best?

iandobbie, 2020-04-02:
I have put the configs into https://github.com/MicronOxford/configs can you put the microscope config for the simulated dsp in there.

iandobbie, 2020-04-02:
I think we should merge this. It is a relatively minor change. Although not ideal it is better than the existing additional window with very limited functionality. We can see if we nee to implement our own status bar class in future.

carandraug, 2020-04-02:
> I have put the configs into https://github.com/MicronOxford/configs can you put the microscope config for the simulated dsp in there.

I will do so.

> I think we should merge this.

I added a few more related changes and merged it.

Closing this with 28ab2bbe96f6b8ccdcfa

closed 2020-04-02 by carandraug