microscope-cockpit/cockpit #566: We need a quick-start guide.

iandobbie, 2020-03-31:
Chris thinks he has cockpit installed and has no idea what to do. We need a quick-start guide. Once installed how do you run cockpit, what do the various windows do? suggested workflows etc...

iandobbie, 2020-03-31:
I have started to write this, but how are we outputting the .rst files into some formatted version. HTML or pdf?

carandraug, 2020-04-05:
> I have started to write this, but how are we outputting the .rst files into some formatted version. HTML or pdf?

The output in html can be generated by running `python3 setup.py build_sphinx`.

carandraug, 2020-04-05:
As part of the quick start guide we spoke about install instructions. I have written the Debian specific instructions on my "wip-install-linux" branch (or see commit [257e6eda](https://github.com/carandraug/cockpit/commit/257e6eda7c30f159537a6967220b25674103e4fe))

iandobbie, 2020-04-15:
I just wrote similar instructions for a blank macos catalina install. I put it directly in the INSTALL file. We also need to deal with the link format as the github readme display wont follow them. My updates are in https://github.com/iandobbie/cockpit/tree/macos-install

iandobbie, 2020-04-15:
My machine didn't have sphinx installed so couldn't build the docs. Do we need to add it to the requirements? I have pulled both your linux instructions and the macos install instructions into the master branch. Needs a bit more editing to make the format correct.

carandraug, 2020-04-15:
> My machine didn't have sphinx installed so couldn't build the docs. Do we need to add it to the requirements?

No, because sphinx is only required to build the documentation which the vast majority of people won't do. We do have a check for its presence on microscope's setup.py which gives a better error message. I can pull something like that for cockpit, what do you think?