microscope-cockpit/cockpit #567: Cockpit windows reduction suggestions.

iandobbie, 2020-03-31:
Related to David's removal of the status light window (#565), we need to do some more interface cleanup. 

My suggestions:

1) Camera views not having blank panels (#549 

2) Relegate logging panel to a secondary window so it isn't always open and active.

3) merge macrostage and mosaic windows, or maybe just move to the touchscreen display. 

These changes would go from 6 windows to 3 and remove a blank panel on the camera panel.

iandobbie, 2020-03-31:
Sorry point 1 should point to #545, silly typo on my behalf. This is fixed in a minor pr #568

juliomateoslangerak, 2020-04-03:
Totally agree with 1 and 2. For the logging window, would it be possible to have something merged into the main window, like the status light, that is either grey when ok and gets red when error. Clicking on it opens the log/ error window.
I'm not sure about merging macrostage and Mosaic. Not trivial anyway to keep functionality and mosaic surface.

iandobbie, 2020-04-05:
How about having logging as a secondary window, therefore hidden by default. Then have it auto show on any error log?

carandraug, 2020-05-07:
Points 1 and 2 are now done (issue #545 and issue #575).

The only suggestion left is the merge of mosaic and macro stage or rework the touchscreen.

iandobbie, 2020-07-08:
Danny is working on the reworked touchscreen which solves some of the issues that David brought up at the group meeting today (we should have invited Julio! I didn't think of it till now). I think we should close this issue now and leave touchscreen updates and merging of the macrostage and mosaic till after that to see how it works on there

closed 2020-07-08 by iandobbie