microscope-cockpit/cockpit #569: "set image size" via camera viewpanel selector does not work

carandraug, 2020-04-02:
On issue #545 @juliomateoslangerak wrote:
> I'm missing something that was accessed through a menu on the camera 'title', that was a quick menu to change ROI. Is this option gone?

Took me a while to find this, but it buttons with the text "Set image size to 512x512" and "Set image size to 256x256" can exist (there would be one entry per possible image size). For this to work, the cockpit camera device needs to implement the `getImageSizes`. If using a Microscope camera it returns an empty list (this comes from the commits c4451f440b92 and 146a3906f907 back when microscopeCamera was still universalCamera).

If you use the cockpit dummy camera, those menus will still appear as you expect.

However, I'm not sure how to make this work for microscope cameras, we don't know ahead of time what sizes are possible. To me this looks like something that should go into the camera panel of the main window (currently this is only done via the camera roi setting).

juliomateoslangerak, 2020-04-03:
I believe this used to be set by some config option. Cockpit side (depot config) I believe.

> To me this looks like something that should go into the camera panel of the main window 
As far as I remember both, the top of the camera window and the camera panel of the main window, were showing the same options... While there is no need to have that option in both places, it was intuitive to have it nexuses to the image...

carandraug, 2020-04-24:
We spoke about this on a meeting two weeks ago and decided to just remove this option since it's not working for a long while and we have not even noticed.

carandraug, 2020-04-25:
Done with 9c34c2cc1f8d . Closing.

closed 2020-04-25 by carandraug