microscope-cockpit/cockpit #570: use of event names should be replaced with the matching variables

carandraug, 2020-04-03:
Events are identified by name (a string).  Using events required knowing the exact string and if there was some typo nothing would happen since it was just assumed there was just no subscriber to that typo-ed event name.

Since 242f399c441e there are global constants in the `cockpit.events` module to prevent this (a typo on the constant name will raise a `NameError`).  However, actually replacing the strings with the constants is an ongoing process.  Commits that have so far done it are:

892b21133a27
f4690fcc4089
3a13249890d5
b73a0db1f813
6a04d1ab7c35
d9ff372386a7

The current state means that when looking for one event we have to grep for both the constant and the string , would be nice to get this done once and for all.

carandraug, 2020-04-04:
I have pushed 1356bedab05ef4fe24d which sed replaces all event names for which there is a matching constant.  While doing that, I found that there were two events that are not used anymore and removed them with be97f82d55744f3e05145f94e9 .

I also found a series of events that do not have a matching event constant, they are:

* cockpit initialization complete
* dummy take image
* filter change
* filter change
* global filter change
* image pixel info
* light exposure update
* load exposure settings
* macro stage xy draw
* mosaic start
* mosaic stop
* new site
* objective change
* program exit
* save exposure settings
* site deleted
* soft safety limit
* stage step index
* stage step size

carandraug, 2020-04-05:
With the removal of the dummy camera and dummy imager, the above mentioned 'dummy take image' has been removed.

iandobbie, 2023-03-08:
I will try to eliminate these. Its relatively simple just takes some work. I'll submit commit numbers with specific fixes.

iandobbie, 2023-03-09:
Events fixed:
 
cockpit initialization complete with 31f921ff3d21ef65480f01b21a3306abb7e0e7af
Filter change with 09ef3722dc6f64e5cbe7ad1dd0cd35fafcc73d70

iandobbie, 2023-03-09:
image pixel info fbbd177079d93c23c643597a7d843c4cc4c29cf3 
light exposure update d2e8f77579b941b695c101f38ee7d2caa6b6f525 
macro stage xy draw 5a5bfda3874ed14d8edd47060ab47866a743742b

iandobbie, 2023-03-10:
Mosaic start and stop f164152826aff9bbd41af49f74951a92a1cac3c9
objective change 54b82cc642098f5344dbdfc3397cc16586037ced
soft safety limits 76b606ce5e7a3f9bad7bee95d89d96196ff4bb7c
Extra change for macro stage xy draw f3d23af8b3c0b453d5cff618fd1e8c176ae3ab9a

iandobbie, 2023-03-10:
Stage step size and index 61b51c6235a89ae299332c5e6307d7c276715cc1

All the events above are now constants, there still appears to be a couple in the test suite

cockpit/testsuite/test_gui.py:        cockpit.events.publish('test gui')
cockpit/testsuite/test_events.py:        cockpit.events.publish('other ' + self.event_name)

These are published but never subscribed as far as I can tell.

iandobbie, 2023-03-15:
Marking as closed as the two remaining calls in the testsuite seem superfluous.

closed 2023-03-15 by iandobbie