microscope-cockpit/cockpit #572: replace cockpit devices wth microscope test devices

carandraug, 2020-04-04:
Cockpit has its own full implementation of dummy devices. Microscope also has them. Since cockpit can make use of Microscope devices, it probably makes sense to just replace Cockpit dummy devices with the test devices in Microscope. This means that next time we want to add a new feature to test/dummy devices, we only have to do them in one place.

carandraug, 2020-04-04:
I have a proposal that replaces the current `dummyCamera` with something that uses Microscope's `TestCamera`. See my `wip-microscope-test-camera` branch or [commit d5e4fce25041](https://github.com/carandraug/cockpit/commit/d5e4fce25041b1eed50bfc6f8ae7ce06c105f6ef)

carandraug, 2020-04-04:
One thing I noticed is that this makes cockpit dependent on Microscope. This is already true, Cockpit needs microscope to make use of microscope cameras even if they are on a remote system. However, it is not declared on `setup.py`. I will assume that it missing from setup.py was a lapse and add it on the next few days unless someone tells me otherwise.

carandraug, 2020-04-05:
@iandobbie wrote:
> I think this is part of the evolution. As Cockpit is now effectively dependant on microscope we should probably move to this and save the repeated code.

carandraug, 2020-04-05:
I have now pushed d9932a1 (mentioned earlier) for dummy cameras and 77942cd  for dummy DSP. The only remaining dummies are the stages (but we don't yet have a test stage on microscope) and objectives (but those won't be microscope devices).

carandraug, 2020-04-15:
Dummy lasers have now been added. Now missing dummy stage, objectives, and SLM.

iandobbie, 2020-04-16:
I am working on a dummy microscope stage, see https://github.com/MicronOxford/microscope/issues/144

iandobbie, 2020-04-21:
initial implementation of a microscope device stage and a cockpit interface are in 

https://github.com/iandobbie/microscope/commit/5145fcba44d7bcffff5c80d2c23327206eb09637

https://github.com/iandobbie/cockpit/commit/6508217ef9ad6175c6bd2c1389f13a77a72db3fa

seems to be working but probably needs cleanup.

carandraug, 2020-04-21:
I took a look at the cockpit `MicroscopeStage`.  Just a few comments:

1. Why `MicroscopeXYStage`? Why not a single cockpit device class that can handle any combination ox xyz?

2. There's an assumption that the axis will be named `'x'` or `'y'` but that's not true. This will need to be on the config.

3. Nowhere is the stage actually enabled. You must call enable to ensure the stage will work. This might make the stage move to its limits.

4. The comment "Limits have to come from config as some stages eg PriorProScanIII can only find limits by moving to them and we dont want to do that every init." has a few issues. The zaber stages won't move at all unless they move to the limits first. They do this as part of their own initialization. Having limits on the config won't save you doing those moves. Also, the microscope stage implementation should already have the logic that prevents it from doing multiple times.

iandobbie, 2020-04-22:
1) yes of course. This was imported from the work I did for the prior stage last year. 

2) yes, it needs some mapping to the axes expected by macrostage etc...I will implment something and see what you think

3) yup, definitely needs an enable call and trap on the microscope side to bypass if already enabled. We might need to add home functionality to the microscope base class

4) comment comes from the prior implementation. I think limits are hardware specific and so should come from the microscope side (as they do in this implementation) so this comment should just go, might need to be mentioned in the prior driver in microscope.

carandraug, 2020-05-05:
I was looking into the microscope stage for cockpit. Are the move callbacks for a PositionerHandler, i.e, `moveAbsolute` and `moveRelative`, meant to block, or should they return as soon as the instruction is sent to the device? I can't find it anything about it (the `stageMover` interface has special code for blocking but the stages themselves, at least picomotor and the PI M687, seem to already do the blocking).

iandobbie, 2020-05-06:
I would be very wary of changing this without good test cases. It was an outstanding question over several iterations with Mick. There are several possible stage hardware implementations. 

1) Controller will accept single or multi axis movement any command at any time.
2) Stage will only accept one axis control at once, but still receives commands when it is moving
3) Stage can accept multi axis movements but will not respond while moving
4) stage accepts only one axis at once and wont respond during movement.

Yes this logic is implemented in several places but I would suggest not fiddling while we don't have hardware to test it with.

carandraug, 2020-05-06:
> Yes this logic is implemented in several places but I would suggest not fiddling while we don't have hardware to test it with.

I'm not fiddling with anything. I'm asking what is the expected behaviour of the `moveAbsolute` and `moveRelative` callbacks so that they can the Microscope stage, which is not yet implemented, can implement the expected behaviour.

carandraug, 2020-05-23:
@iandobbie I picked on your implementation and generalized it for the other axis. Can you take a look at my `wip-572-microscope-stage` branch? The deviceserver configuration I used is:

```
from microscope.devices import device, AxisLimits
import microscope.testsuite.devices as testdevices

DEVICES = [
    device(testdevices.TestStage, '127.0.0.1', 8008,
           {'limits': {'X': AxisLimits(0, 25000), 'Y': AxisLimits(0, 12000)}}),
    device(testdevices.TestStage, '127.0.0.1', 8009,
           {'limits': {'Z': AxisLimits(0, 25000)}}),
    device(testdevices.TestStage, '127.0.0.1', 8010,
           {'limits': {'Z': AxisLimits(0, 2500)}}),
]
```

which matches the following on my `depot.conf`:

```
[XY stage]
type: cockpit.devices.microscopeDevice.MicroscopeStage
uri: PYRO:TestStage@127.0.0.1:8008
x-axis-name: X
y-axis-name: Y
x-units-per-µm: 1
y-units-per-µm: 1

[innerZ]
type: cockpit.devices.microscopeDevice.MicroscopeStage
uri: PYRO:TestStage@127.0.0.1:8010
z-axis-name: Z
z-units-per-µm: 10

[outerZ]
type: cockpit.devices.microscopeDevice.MicroscopeStage
uri: PYRO:TestStage@127.0.0.1:8009
z-axis-name: Z
z-units-per-µm: 1
```

It's missing the code to compute the movement time, which I'm not sure how to handle.

carandraug, 2020-05-26:
Before this issue becomes about discussing the implementation of support for microscope stages instead of just the dummy devices, I have opened issue #614. I'm marking this issue as blocked by it.

carandraug, 2020-07-10:
Now that we have microscope stages in cockpit, I'm replacing the cockpit dummy stages with the microscope test stages. Proposal on my "wip-572-dummy-microscope-stages" branch.

carandraug, 2020-08-28:
No comments after 7 weeks. Rebased and merged the last proposal. Closing as fixed.

closed 2020-08-28 by carandraug