microscope-cockpit/cockpit #573: can we remove the softCamera device module?

carandraug, 2020-04-04:
The `cockpit.devices.softCamera` modeule "makes software-driven cameras available to cockpit." Reading though it it seems dependent on some remote implementations which I don't know where they are. It's suggested that it offers support for flycap2 and picam. Can we remove this?

iandobbie, 2020-04-05:
We ought to fully support the picam as the support is mostly there and it could be a useful cheap a cheerful approach for getting a simple setup working. I brought home a red pitaya with the idea that I would get the hardware trigger working in both the Ximea and the picam code. Havent even got the cables I need to trigger the cameras built yet, but shouldn't be difficult.

mickp, 2020-04-05:
Also, if Diamond are still using cockpit on their AxioImager for overview
imaging, they'll need the SoftCamera to support FlyCap2.

carandraug, 2020-04-06:
We just spoke on the lab meeting and we are removing this.

About picam, @iandobbie  has a picam implementation for microscope so that is preferred.

About flycap2, seems like Diamond is not using it anymore. This would still remove support for the FlyCap2 if someone later is interested on it but if that happens we should then be adding it to microscope. For future reference, we believe the remote being used was [mickp's FlyCapture2-ctypes](https://github.com/mickp/FlyCapture2-ctypes).

carandraug, 2020-04-06:
Now done with d7b1d52

closed 2020-04-06 by carandraug