microscope-cockpit/cockpit #574: Default windows positions are mad and maybe should be totally removed

iandobbie, 2020-04-06:
As can be seen from Chris's no mosaic issue we need to change or totally remove the default window positions. Chris couldn't get the mosaic on screen as its window position was off the screen and the move to mouse functionality was pretty opaque.

carandraug, 2020-04-06:
So Chris (does Chris have a github account?) issue is that the default mosaic window position is at x coordinates 1280, which on his case would be outside the screen. This is the first time someone runs cockpit and there's no `config.py`.

It's interesting to note that the behaviour is backend specific.  On my case, with GTK3 backend, if the window is positioned outside the display it will be moved into its closest valid position.  But even if the issue is windows specific, I agree it's crazy to hardcode window positions now that we are free to do so (I guess that this was from the time when the plan was to require users had a specific number of monitors and graphics card).

With 27e6d407978bd486a I have now removed all hardcoded window positions and sizes.  Two things came out of that:

1. the mosaic window is not large enough to show all the buttons on the left hand side panel.  This is very likely something to do with the window sizer not being set properly.

2. Cockpit's config.py saves the window location but not the window size.  Maybe we want to also save/load them?

iandobbie, 2020-04-06:
1) Not sure about default positions. Are things visible if you just startup with no positions? Are the sizes sensible. We might need default sizes to avoid issues like the mosaic one above?

2) Yes we should store user defined positions and sizes.

iandobbie, 2020-04-06:
Overall a good idea but needs additional modifications.

1) some windows need either a minimum size or a fixed default size.
a) mosaic is totally broken
b) python shell window is too small
c) touchscreen is probably too big. 

we need to either not read positions with --no-config-files or we need to have an extension

iandobbie, 2020-04-06:
Oh, I also managed to start up with "--no-config-files" and have all the windows off my screen and could only recover them by connecting an external monitor. Very frustrating.

carandraug, 2020-04-06:
> a) mosaic is totally broken

Now fixed with 60db148d, e544c15f, and 941bb32b.

> b) python shell window is too small

Probably. I have no clue what would be a good default size, but what if we remove it and replace it with wx's ShellFrame? It has a whole menu with more powerful options. We lose the drag and drop of mrc files that are automatically bound to a variable but we gain a lot more features (and we can still drag and drop files to get their path in the shell). Take a look at my "wip-pycrust-shell" branch.

> c) touchscreen is probably too big

Not sure. It was previously hardcoded for a size 1800x1000. Now its size is based on the size of its controls. It looks reasonably sized too me (actually, I think it looks pretty much the same size).

> we need to either not read positions with --no-config-files or we need to have an extension

This is a separate issue, so I have opened issue #577 for it.

carandraug, 2020-04-16:
> Probably. I have no clue what would be a good default size, but what if we remove it and replace it with wx's ShellFrame? It has a whole menu with more powerful options. We lose the drag and drop of [...]

Any feedback on that?

---

I have since found a few more places in cockpit with hardcoded sizes and will see about removing them too.

mickp, 2020-04-16:
You could drag and drop Mrc files?

iandobbie, 2020-04-16:
Yes, and a whole lot of other things. There was a nice colour viewer you could pull up and stuff. I should pull out my notes I guess.

carandraug, 2020-04-24:
> Yes, and a whole lot of other things. There was a nice colour viewer you could pull up and stuff. I should pull out my notes I guess.

Looking at the code, the only feature Cockpit adds to the python shell itself is the drag and drop of mrc files. Anything else must come straight from wx's PyShell. It is also possible to keep the drag and drop of mrc files. From Mick's surprise I'd guess not used so I think we could just use the whole PyShell Window.

>> c) touchscreen is probably too big
>
> Not sure. It was previously hardcoded for a size 1800x1000. Now its size is based on the size of its controls. It looks reasonably sized too me (actually, I think it looks pretty much the same size).

I found another place where the touchscreen window size is being set, now hardcoded to 1500x1000.

carandraug, 2024-06-08:
> Looking at the code, the only feature Cockpit adds to the python shell itself is the drag and drop of mrc files. Anything else must come straight from wx's PyShell. It is also possible to keep the [...]

This was done in 8829235d (4 years ago).

All other issues described here seem to have fixed (there is possibly more hardcoded sizes in cockpit that need to be removed but there is no list or central place listing them).

Closing as fixed.

closed 2024-06-08 by carandraug