microscope-cockpit/cockpit #576: Save top and bottom dont update Z display.

iandobbie, 2020-04-06:
The saved top and bottom positions need to force a redraw even on the Z axis disolay on both the macrostage window and the touchscreen window.

iandobbie, 2020-04-06:
Not ideal but I have a solution in https://github.com/iandobbie/cockpit/tree/savetopbottom. Has a rather horrid direct calls to the two macroStageZ instances. Not sure how to do this better.

carandraug, 2020-04-06:
Is this the same issue as issue #421 ?

iandobbie, 2020-04-07:
Yes, sorry missed that. I did search but obviously not enough. Although the issue has been there for ages it hasn't been solved. My solution, although rather a hack is a small change and does fix the issue.

carandraug, 2020-04-13:
The suggested fix would mean that the savetopbottom needs to track all instances of MacroStageZ. It is also directly modifying the internals of the other windows which it shouldn't, it can break easily.

I'm closing this as duplicate of issue #421  to discuss alternatives.

closed 2020-04-13 by carandraug