microscope-cockpit/cockpit #578: DPI awareness on Windows

dstoychev, 2020-04-07:
If the display scale on Windows is changed, ``cockpit`` is unaffected and this could produce windows that are too large or too small. There should be a way to preserve the original intended size of the user interface elements regardless of the DPI/scaling chosen.

iandobbie, 2020-04-07:
You said in the group meeting there was an easy fix. What is this?

mickp, 2020-04-07:
This is at odds with using native UI elements, and if you go over-riding the system-wide DPI setting, you could end up with unreadable text.

dstoychev, 2020-04-07:
Unfortunately, without DPI awareness cockpit does not work on the Surface Laptop and other similar devices. The problem is that there are situations where the screen may be too small to accommodate some of the windows. Case in point: I have no issues on the Surface Studio (28" screen, ~192dpi). On the Surface Laptop (13.5" screen, ~200dpi) some windows barely fit and the touchscreen does not fit at all. The DPI of the two devices is comparable and I use 150% scaling on both of them, but the laptop has only quarter (!!!) of the screen area. 

@iandobbie, I used the following code to make the software DPI aware:
```
if os.name in ('nt', 'ce'):
    from ctypes import OleDLL
    OleDLL("shcore").SetProcessDpiAwareness(1)
```
However, David rightfully pointed out that there must be a way to do this via wxWidgets instead of calling windows API directly.

carandraug, 2021-01-29:
We recently pushed a fix to handle the Mac OS retina. Part of that fix is to automatically adjust the size when the window moves between displays with different DPIs. @dstoychev is this still an issue or can it be closed?

dstoychev, 2021-01-30:
Honestly, I do not remember what this is about and reading my comments above I cannot understand what the issue is. At the moment, both with and without (version 2.9.1, commit bcb3467fe6957fa16a92010cd29aef30b4a8633a) the recent DPI changes the UI scales accordingly. My hack from above doesn't make any difference either. I suspect Windows may have done something on their side in one of the recent updates. In any case, closing this.

closed 2021-01-30 by dstoychev