microscope-cockpit/cockpit #579: Most important issues.

iandobbie, 2020-04-07:
Just thought it would be good to list the most important issues to try to iron out before publication.

Please feel free to add comments, or additional issues, or even better close the open issues so we can delete them here!

1) Window reduction - ongoing see issue #567 
2) startup option, either touchscreen or mosaic+macrostage - no issue yet (I think)
3) clean up configs and initial startup - issues #572 #577 #574 and others I am sure
4) documentation - eg issue #566 
5) various nagging issues from ages ago - #19, #22, #179, #217, #284, #391, #429 

Additionally I am sure there are issues that have been solved but just need to be tested and closed.
#5, #256, #282, #306, #390, #409,  #412, #413, #424, #438, #438, #443, #448, #461, #463, #466, #472, #473, #492, #493, #496, #506, #507, #510, #517, #526, #530, #536, #545, #558, #559, #561, #565 

This is from a quick scan of the list. If anyone can take a couple and see if they can close them that would be great.

carandraug, 2020-05-05:
Instead of an issue to list the most important issues, I think we could instead use the labels. We already have the labels "publication-blocker" and "critical" for the important ones and "needs testing" for the others.

So, changing this issue aim a bit, what other labels should we have? Maybe one for UI/UX looks (improve style but not actual something that is broken)?

iandobbie, 2020-05-05:
Yes probably worth doing, but someone is going to hgave to go through the issues and sort them properly into these sections.

carandraug, 2020-05-05:
> Yes probably worth doing, but someone is going to hgave to go through the issues and sort them properly into these sections.

For the most important and the ones that need testing you already did. I think for the others we just do them as we interact with them, no need to go back and classify them.

From the top of my head, here's some labels I think we could use:

* hardware: an issue associated with a specific piece of hardware, i.e., it can't be replicated with normal dummy devices
* critical: an issue that requires immediate attention. This is for issues that cause cockpit to crash or a major loss of function in any platform.
* enhancement: new feature request but possibly performance enhancement as well
* question: not a bug, someone asking for help
* blocked by other: fixing this issue requires, or would be much easier after, fixing some other issue
* GUI: some issue on the actual graphical user interface. It can be labelled together with some OS specific label or with enhancement.
* patch provided: someone has sent a patch or a pull request that needs to be reviewed and merged
* Windows: specific to the Windows OS
* MacOSX: specific to MacOSX
* Linux: specific to Linux

Also the following labels can be used when closing an issue:

* duplicate: self-explanatory, the comment closing should link to the original issue.
* invalid: either because it's not our bug, it's the intended behaviour, or it was an user error in the first place.

And I propose we remove these labels:

* bug: instead of this label, let's just assume that it is a bug unless it has a label saying otherwise
* publication-blocker: I have mixed feelings about this. It does not really belong to the cockpit repo, maybe keep this on the paper repo.
* needs-testing: redundant with "patch provided"
* wontdo: I think this is a bit redundant with invalid.

Nothing stops us from adding or removing labels in the future, this is an initial proposal. If you think it looks good, I can add them to the documentation under a new "Maintainers" section.

iandobbie, 2020-05-11:
Please do go ahead with this. We need a defined system for when we get more users or contributors. We can easily update the plan if we find it lacking in some aspect.

carandraug, 2020-05-11:
> Please do go ahead with this.

Now done. I did not add it to the documentation since I found that we can have the description on github labes page itself, so we can just link to https://github.com/MicronOxford/cockpit/issues/labels instead.

Closing this meta-issue as fixed.

closed 2020-05-11 by carandraug