microscope-cockpit/cockpit #580: Suggested simple GUI updates.

iandobbie, 2020-04-08:
Suggested minor GUI updates from lab meeting 2020-04-08
Main cockpit window

1) Objective dropdown has different corners to the other buttons on interface.

2) Labels of buttons have variable case, some start with capitals, others don't. 

3) Shorten label of "Dummy camera 1" etc.

4) Disabled devices should be grey and not black

5) If there are real light sources disable the ambient one (maybe, or make it a config option)

iandobbie, 2020-04-08:
Turns out the status bar is obscuring the exposure time boxes of the light sources. Can we increase the

iandobbie, 2020-04-08:
I have a new branch GUI-fixes which fix all issues above except 5.

carandraug, 2020-04-13:
> Turns out the status bar is obscuring the exposure time boxes of the light sources. Can we increase the size of the panel so the status bar doesn't obscure it.

The problem is that the frame fitting was done before adding the status bar. This is now fixed with d31c1126536c5

> Objective dropdown has different corners to the other buttons on interface

I took a look at this and I think it depends on how you look at it. The objective dropdown menu is in a panel with raised border (I think that's what you mean with corners) while the buttons on the top of the window are not. However, all the other panels (cameras, filters, and slm) are in a panel with raised border. I believe the logic here is that the objective panel should be in the same style as the other device panels which it does. The commit that added raised border around the objective is 966c48e81cc6539af8df .

And whether the objective panels should be like the buttons because it's on the top row instead of the middle of the other devices, that is dependent on whether there's only one objective device since 43fc89aa0276f2c

> Labels of buttons have variable case, some start with capitals, others don't.

I have pulled and pushed your commits with a modified commit message to point to this issue.

> Shorten label of "Dummy camera 1" etc.

Sure, but I don't think the proposed "Vcam" is clear what it means. Note that the name "Dummy camera" only appears if there's no devices so it's either for developers (who won't care) or people who did so accidentally and in which case may be confused with "Vcam". I don't think the length of the dummy camera device affects much the general UI.

> Disabled devices should be grey and not black

Why grey and not black? What does the grey colour reveals that the black colour does not?

> If there are real light sources disable the ambient one (maybe, or make it a config option)

I think this is easy. We can just remove it. If someone wants it, they can add it to depot conf like any other device. And like the other dummy devices, we can add a dummy laser if there's no light sources. Sounds good?

iandobbie, 2020-04-14:
> > Objective dropdown has different corners to the other buttons on interface
> 
> I took a look at this and I think it depends on how you look at it. 

Yes of course. I later realised that it was the border and my change added a few pixels of padding making it look slightly less out of place. 

https://github.com/iandobbie/cockpit/commit/692d0f05ad05de41c866b1f94cdf1b3b48de782f

> Sure, but I don't think the proposed "Vcam" is clear what it means. Note that the name "Dummy camera" only appears if there's no devices so it's either for developers (who won't care) or people who [...]

Good point. Happy to leave as is but Danny will have to think about the touchscreen interface. How about a slight shortening. "DummyCam1" 9 rather than 14 chars?

> > Disabled devices should be grey and not black
> 
> Why grey and not black? What does the grey colour reveals that the black colour does not?

Nothing, just as aesthetic point made in the discussion. I am not wedded to either but if people are happier with one rather than the other then might as well change it.

> I think this is easy. We can just remove it. If someone wants it, they can add it to depot conf like any other device. And like the other dummy devices, we can add a dummy laser if there's no light [...]

Yes, something like that would be good.

carandraug, 2020-04-15:
> Yes, something like that would be good.

I have now pushed a series of commits that do this. Added a new DummyLaser device class, automatically instantiates a series of dummy lasers, creates an AmbientLight device class, and removes ambient from being started automatically requiring it to be manually added by the user on depot.

carandraug, 2020-04-15:
> Good point. Happy to leave as is but Danny will have to think about the touchscreen interface. How about a slight shortening. "DummyCam1" 9 rather than 14 chars?

The name of cameras in a real situation is set by the user, not by cockpit. If @dstoychev remake of the touchscreen interface has a limit on the length of the name, he already has to think about the touchscreen interface. @dstoychev can you open an issue (or possibly multiple if you believe it's better) describing the aim of this touchscreen remake to discuss whatever comes up with it?

carandraug, 2020-04-16:
>>> Objective dropdown has different corners to the other buttons on interface
>>
>> I took a look at this and I think it depends on how you look at it.
>
> Yes of course. I later realised that it was the border and my change added a few pixels of padding making it look slightly less out of place.
>
> iandobbie@692d0f0

I'd prefer avoiding insertion of non-relative lengths and spacers, we are just working on removing them as part of issue #574 .

If the root of the problem is the raised border looking odd, what about removing them and delimit the panels with something else? I think a [wxStaticBox](https://docs.wxwidgets.org/3.0/classwx_static_box.html) could be just the thing but I have historically had problems with them when working on the experiment panel. At the time I instead used a `StaticText` and `StaticLine` and had a line only at the top instead of all around, which seems pretty clean. When I tried it on our main window, it looks like this (under GTK3):

[image]

While at the moment it looks like this:

[image]

What do you think?

iandobbie, 2020-04-16:
I actually think the line looks better but we ought to ask around a bit.

carandraug, 2020-08-27:
> Objective dropdown has different corners to the other buttons on interface.

Since 7e08699370b0c18c6025 the objectives panel no longer has different corners. The different corners came because the single drop down menu had extra borders like all the other device panels since there were cases where it could have multiple drop down menus. This has now changed and it always has a single one and the raised borders were removed.

carandraug, 2020-08-28:
Of the original suggestions. I think they've all been addressed or there's no special reason to be addressed, so I'm  closing this.

> 1. Objective dropdown has different corners to the other buttons on interface.

Done as a side effect of refactoring objectives.

> 2. Labels of buttons have variable case, some start with capitals, others don't.

Done.

> 3. Shorten label of "Dummy camera 1" etc.

This is only used for our development, the users will name them whatever they want, so nothing to do here.

> 4. Disabled devices should be grey and not black

No clear reason why this improves anything, so let's not doing that until then.

> 5. If there are real light sources disable the ambient one (maybe, or make it a config option)

Done, ambient light is a separate device and users can add it to the depot file if they want.

closed 2020-08-28 by carandraug