microscope-cockpit/cockpit #582: LightPowerHandler colour type consistency

dstoychev, 2020-04-16:
In the [``getHandlers()``](https://github.com/MicronOxford/cockpit/blob/d2b7998d63a6581d523c4af074098c9b39380ae4/cockpit/devices/microscopeDevice.py#L300) method of the [``MicroscopeLaser``](https://github.com/MicronOxford/cockpit/blob/d2b7998d63a6581d523c4af074098c9b39380ae4/cockpit/devices/microscopeDevice.py#L279) class, the colour is set in the following way:

```
wl = self.config.get('wavelength', None)
if wl:
    col = cockpit.util.colors.wavelengthToColor(wl, 0.8)
else:
    col = '0xaaaaaa'
```

The problem is the inconsistency in the type of the ``col`` variable, which is either a tuple (returned by ``cockpit.util.colors.wavelengthToColor()``) or a string. I see two solutions: the else statement could be changed to a tuple or the an error could be raised if the wavelength attribute is not set in the config. I think I prefer the latter, but I am not sure how well it fits with the overall config handling in the package.

carandraug, 2020-04-16:
That `col` argument is only used to construct a `LightPowerHandler` which then no longer uses it. Digging through the history, it was used by the old illumination controls which were removed with c8b5969f3048b843362c . So I propose a third solution of just removing those lines, modifying `LightPowerHandler.__init__` to no longer take a color argument, and fix all other constructions of `LightPowerHandler`s.

You may find other bits of the `LightPowerHandler` that are no longer used and should be removed. For example, the `setPowerArbitrary` method can also be removed (it was also part of the old illumination controls).

carandraug, 2020-04-24:
With 3631d511cde1 I have removed the color property from the `LightPowerHandler` as well as a few other unused properties, methods, and imports, leftovers from the old controls. I have done the same on `LightSourcehandler`.

Closing as fixed.

closed 2020-04-24 by carandraug