microscope-cockpit/cockpit #583: discoverability of list of windows and menu bar on main window

carandraug, 2020-04-16:
One can right click on the background of the main window (and view of a disabled camera, and even parts of other windows) to get a menu listing all the windows with the option of bringing them to the front and moving them to the mouse position. However, we spoken before that right click on the background of a window is not easy to discover and we could instead create a menu bar on the main window which is much more standard.

iandobbie, 2020-04-16:
I think the menu bar idea is much more the expected behaviour. The right click on background is ok once you know its there, and so long as you haven't also lost the main window which I have done at times.

carandraug, 2020-04-17:
I looked at moving the windows menu into the menu bar and found an issue. The menu bar is added when the window is created, which means the menu must exist by then. However, for the menu to be created the windows themselves must already exist because the menu is created dynamically based on the windows that already exist.

This all stems from the fact that it's the App `OnInt` method that creates all the windows. This is odd because windows are children of the main window so it makes sense to be the main window to create its children. Unless someone can remember a reason not to do it, I will investigate that approach.

Also, we spoke last week during a meeting that with a menu bar we could have other common menu items, such as About that link to the cockpit website and documentation, and a quit. The default quit menu item would come with a shortcut. To prevent accidental exiting of cockpit, I'm thinking of adding an "Are you sure?" modal dialog to it.

carandraug, 2020-04-29:
I have added an initial menu bar with:

1. File menu with Quit entry
2. Help menu with About entry and a link to source code

It's on my "wip-menubar" branch for someone to take a look. If it looks good, I would push them before working on having a Windows entry on the menu bar.

carandraug, 2020-05-07:
I have pushed the above proposed changes, which adds a menu bar to the main window. We now need to add a windows menu to the menu bar. For that to work, I think we should be moving the construction of the children windows into the construction of the MainWindow itself.

carandraug, 2020-06-05:
I now have a branch that moves the windows menu to the widow menu bar. Can someone test my `583-widows-menu` branch for a ride? I kept the structure of the menu the same, the only practical differences is the change of shortcut from Ctrl+P to Alt+W (Alt is the standard shortcut to open the menu bar and W because of **W**indows).

> We now need to add a windows menu to the menu bar. For that to work, I think we should be moving the construction of the children windows into the construction of the MainWindow itself.

Turns out that this was not required. Instead, each time the menu is open it checks for new windows. This might cause issues if a window ever stops existing (I don't think this ever happens at the moment though).

iandobbie, 2020-06-08:
Definite problem with the logging panels.  Screenshot of this running on a mac

[image]

carandraug, 2020-06-08:
> Definite problem with the logging panels.

That issue is not because of the changes in the proposed changes. It is also reported on #618

iandobbie, 2020-06-08:
Found another issue that this changes the behaviour of menubar:python:quit cockpit. Previously this menu selection (or the associated hotkey) quick cockpit as expected. Now it just closes the main control window. This is all on  a mac

carandraug, 2020-06-10:
> Found another issue that this changes the behaviour of menubar:python:quit cockpit. Previously this menu selection (or the associated hotkey) quit cockpit as expected. Now it just closes the main control window. This is all on a mac

I'm not sure what menu bar item you're talking about. On the main window menu bar there is only 'File>Quit' and no 'Python:quit'. If that menu item no longer exits cockpit, are you sure that is related with the proposed fix and does not exist currently on master? Because I can't see how the proposed fix would affect that.

iandobbie, 2020-06-11:
On a mac the quit option is under the "Python" menu, which is the first entry in the menu bar not under "File", hence the comment. I just tried to test this further but the branch doesn't seem to exist in your repository any more.

carandraug, 2020-06-11:
> On a mac the quit option is under the "Python" menu, which is the first entry in the menu bar not under "File", hence the comment. I just tried to test this further but the branch doesn't seem to exist in your repository any more.

The branch is still there, it is named `583-windows-menu` and has not been merged yet. Or are you still trying to test `wip-menubar`? That was more than 1 month ago and it has since been merged.

iandobbie, 2020-06-11:
I think I am braindead today. I don't know why but I could seem to checkout the branch this morning and now I can. My issue with the menubar quit item not closing the program seems fixed. However the logging window placement still makes it unusable on the mac .

[image]

carandraug, 2020-06-12:
> [...] My issue with the menubar quit item not closing the program seems fixed.

I have tried this on a Mac with @ARCFaria and I now understand what you mean by the "Python" menu. Macs move the frame menu bar to the bar at the top of the monitor, add an extra menu with the program name ("Python" in this case, but how to make it appear as "Cockpit"?), and then moves the Quit and Help menu items in there. This is all new to me and it is happening automatically which is why I had no idea what you meant by the "Python" menu.

Anyway, the problem is roughly explained on wxwidgets wiki [WxMac-specific_topics#Single_menu_bar](https://wiki.wxwidgets.org/WxMac-specific_topics#Single_menu_bar):

> wxMac handles this automatically by displaying the menu bar associated with each wxFrame at the top of the screen whenever that frame is in front. If your program has multiple frames and the user brings a different frame to the front, that frame's menu bar will be switched in.
>
> Warning: be careful of modeless windows which do not have menu bars. (Modal dialogs are okay.) If a modeless window gets brought to the front and it doesn't have a menu bar, the last menu bar will stay at the top of the screen. When the user selects something from one of these menus, the selection might never get handled, since the active frame doesn't recognize the command. You have two choices: either handle your commands in your wxApp subclass (since the commands will eventually get sent there), or create a degenerate wxMenuBar for all of your frames (maybe with only a Quit item). 

In the case of cockpit, only the main window and the python shell window have menu bar. @iandobbie, I believe what happened was that you currently had the python shell window opened, so when you did "Python:Quit" you interacted with the python shell window menu bar instead of the main window menubar. Not being a Mac user, I'm not sure what sort of behaviour and interface we should have here. What would make sense?

iandobbie, 2020-06-12:
I think "quit" should always quit the application, maybe with an "are you sure" dialog. I will investigate how different the main menubar and the python shell menu bars are.

iandobbie, 2020-06-12:
Ah, I didn't quite realise the complexity of this! On the mac it appears as though all the windows except the pyshell window give the same menubar and the quit functionality works on them. Switching to the pyshell gives a different set of menubar menu's which is a little disconcerting when you first notice it but I think is fine overall. I see two issue which I think would help a lot.

1) make the quit option (on a mac this is menubar:python:"quit cockpit") actually quite cockpit not just hide the pyshell window

2) would it be possible to add the windows menu to the pyshell menubar? This would make it much easier to return to the main window or other window from the pyshell.

carandraug, 2020-06-12:
> 1.make the quit option (on a mac this is menubar:python:"quit cockpit") actually quite cockpit not just hide the pyshell window

The quit menu item you're talking about is only when the pyshell frame is the active frame. In all other platforms, this menu item will appear on the pyshell frame in the Files menu so it's very clear that we're exiting the python shell program. I would find it quite weird that it closed cockpit. Also, note that this problem won't be unique to pyshell, any other frame with a menu bar will have the same problem. At the moment, the only other such frame is the value log viewer.

Also, note that the menu item on each of these menu bars is a different object, it's not the same menu item shared between them. I guess technically we could hijack the exit event to force cockpit and limit that hikacking to Macs only.

> 2. would it be possible to add the windows menu to the pyshell menubar? This would make it much easier to return to the main window or other window from the pyshell.

It would be a hack but technically it is possible. But I don't think it fits though, it should be the main window that controls its program window positions.

Maybe if instead of "Python" the menu said "Cockpit" and "PyShell" on mac this would make more sense. I have no idea how to do it though.

---

Anyway, it seems there is nothing specifically against the patch that moves the windows menu from right click to the menu bar, so I pushed it 3fbc47a95894.

iandobbie, 2020-06-15:
> The quit menu item you're talking about is only when the pyshell frame is the active frame. In all other platforms, this menu item will appear on the pyshell frame in the Files menu so it's very [...]

Except it quite clearly says "Quit cockpit" so I dont think you would find it weird. 

> Anyway, it seems there is nothing specifically against the patch that moves the windows menu from right click to the menu bar, so I pushed it [3fbc47a](https://github.com/MicronOxford/cockpit/commit/3fbc47a95894f6c0258ca8f49bada3fc54bda553).

No I think overall its a big improvement. I was just pointing out some weirdness on the mac and what I think is an additional improvement.

carandraug, 2020-06-16:
> Except it quite clearly says "Quit cockpit" so I dont think you would find it weird.

I see. The text "quit cockpit" is wx magic. The menu item only gets `wx.ID_EXIT` so it does whatever is meant to look natural in the system. Maybe there is a way to tell wx that the PyShell frame quits a separate program, but I don't know how.

> No I think overall its a big improvement. I was just pointing out some weirdness on the mac and what I think is an additional improvement.

I'm starting to feel that those weirdnesses should go on a separate issue. The behaviour is surprising and seems like it could be improved but I'm not sure how. If you agree, please close this issue.

iandobbie, 2022-07-14:
I agree and will close this issue and open a new mac menu bar issue.

closed 2022-07-14 by iandobbie