microscope-cockpit/cockpit #584: what can cause blocking threads in video mode?

carandraug, 2020-04-20:
I'm trying to understand the need for `clearOneShotSubscribers` at the end of `Imager.videoMode`. The code says:

```
        # Our thread could be blocked waiting for an image.
        # Clear one shot new image subscribers to make sure it
        # is unblocked.
```

I found it was added the first time in 44d21e9b7fbd in `Imager.stopVideo` and later moved to `Imager.videoMode` in 35d30a6e5187. I guess the "thread" the comment refers to is not the `videoMode` function thread since that comment came unmodified from `stopVideo`. Still I can't figure out the situation that would give rise to its need. Anyone remembers what caused this?

iandobbie, 2020-04-20:
Sorry this was all Mick. I haven't done any more than very superficial work on the timing or triggering work

mickp, 2020-04-20:
Can't see the code right now, but probably something like a one shot
subscription gets set up to receive data, but the camera never gets
triggered due to an abort event, or data never comes back due to a camera
problem. Next time video mode runs, there's two events set up to respond to
the incoming data: I think both would run, which means you'd fire off two
of the same local calls which could get in each others' way.

mickp, 2020-04-20:
I think this also applies to the mosiac thread (or any other thread that
blocks until it receives data.). At some point, there were definitely
issues with restarting the mosaic after a dropped frame (which we used to
see a lot due to the Ixon driver/firmware bug regarding time calculations.)

carandraug, 2020-04-20:
Hey Mick! Thank you for the clarification.  I've tried yesterday for some time causing failures in many different places to force the issue but still haven't found it.

> [...] due to an abort event [...]

Abort events should already be clearing up all one shot subscriptions. There is a global subscription to abort events that clears all one shot subscriptions. 

> [...] or data never comes back due to a camera problem. [...]

If the camera problems means that the new image event never gets to happen, then executeAndWaitFor does not return and we never even get to the clearOneShotSubscribers. If the camera problem means that the function raises an exception, then the subscribed function is removed from the list of subscribers before being called, so there will be nothing to clear.

> [...] after a dropped frame [...]

I guess the best place to mock this behaviour in a device independent manner is on `ImageHandler.takeImage`, right? What should it do there to replicate it?

mickp, 2020-04-20:
> If the camera problems means that the new image event never gets to happen, then executeAndWaitFor does not return and we never even get to the clearOneShotSubscribers

It never returns, but clearOneShotSubscribers is called the *next* time videoMode is started, or an abort event happens: on executeAndWaitFor, the the subscriber has an __abort__ attribute set to a method that releases the lock. The abort event or starting videoMode calls clearOneShotSubscribers, which calls the __abort__ method, which releases the lock.

> ImageHandler.takeImage, right? What should it do there to replicate it?

To simulate a hardware problem, it should ignore call, I think: do nothing; generate no data.

carandraug, 2020-04-21:
>> ImageHandler.takeImage, right? What should it do there to replicate it?
>
> To simulate a hardware problem, it should ignore call, I think: do
> nothing; generate no data.

Cool, so I now patched `ImagerHander.takeImager` to simply return the
20th time it is called.  After 19 images, video mode hangs.

>> If the camera problems means that the new image event never gets to
>> happen, then executeAndWaitFor does not return and we never even
>> get to the clearOneShotSubscribers
>
> It never returns, but `clearOneShotSubscribers` is called the *next*
> time videoMode is started, [...]

The next time `Imager.videoMode` is called the following happens (done
by untoggling and toggling the "Video Mode" button):

1. `amInVideoMode` is True so `videoMode()` calls `stopVideo()`
2. `stopVideo()` sets `shouldStopVideoMode` and returns
3. then `videoMode()` returns and does nothing else
4. each time `videoMode()` is called again it goes back to step 1
  because even though `shouldStopVideoMode` is now True, that is
  only checked after `amInVideoMode`

> or an abort event happens: on executeAndWaitFor, the the subscriber
> has an __abort__ attribute set to a method that releases the
> lock. The abort event or starting videoMode calls
> clearOneShotSubscribers, which calls the __abort__ method, which
> releases the lock.

Pressing "abort" releases the thread waiting in
`executeAndWaitFor`. However, this comes because an abort event clears
all one shot subscribers and not because of the
`clearOneShotSubscribers` at the end of `videoMode` which at this
point has not been called.

Another method to break of the hanged video mode is to just "Snap"
image which will publish a new image event which unblocks the video
mode thread which can then end.  However, even in that situation, the
`clearOneShotSubscribers` at the end is called too late.

carandraug, 2020-04-30:
I have a proposed fix on my "wip-videomode-block-584" branch. It basically adds a timeout while waiting for the new image event, a timeout of :

```
            timeout = 2.0 * ((camera.getExposureTime()
                              + camera.getTimeBetweenExposures()) / 1000)
```

If the timeout is reached, the wait is unblocked, an exception is raised, and imager leaves video mode.

mickp, 2020-04-30:
You might want to set a lower limit on this: image processing in software
(e.g. reorientation in python, or byte unpacking where libraries use tricks
to improve camera-PC transfer rates) and network transfer could be the rate
limiting factor for fast, large-frame cameras.

carandraug, 2020-05-01:
@mickp wrote:
> You might want to set a lower limit on this: image processing in software
> (e.g. reorientation in python, or byte unpacking where libraries use tricks
> to improve camera-PC transfer rates) and network transfer could be the rate
> limiting factor for fast, large-frame cameras.

Good point. What do you think of just adding 5 seconds to the time above? It should be more than enough for those steps? It may be even too long but we're only talking about the max time to wait for so shouldn't be a problem.

carandraug, 2020-05-04:
Now merged 25f9b8c2868c . In the case of dropped frames, it now exits video mode on its own after 5 seconds extra. This fixes the hang in the case of dropped frames.

For future reference, here's a patch to replicate dropped frames with the default dummy devices:

```
--- a/cockpit/handlers/imager.py
+++ b/cockpit/handlers/imager.py
@@ -68,7 +68,19 @@ class ImagerHandler(deviceHandler.DeviceHandler):
     def __init__(self, name, groupName, callbacks):
         deviceHandler.DeviceHandler.__init__(self, name, groupName, False, 
                 callbacks, depot.IMAGER)
+        self._imager_to_ignore = 'Dummy camera 1 imager'
+        self._count = 0
+        self._frames_to_drop = [20, 25, 40] # drop these frames
 
 
     def takeImage(self):
-        self.callbacks['takeImage']()
+        if self.name == self._imager_to_ignore:
+            print('acquiring frame #', self._count)
+            self._count += 1
+            if self._count in self._frames_to_drop:
+                print('dropping frame #', self._count)
+                pass
+            else:
+                self.callbacks['takeImage']()
+        else:
+            self.callbacks['takeImage']()
```

closed 2020-05-04 by carandraug