microscope-cockpit/cockpit #585: MacrostageXY display doesn't deal with different sized stages.

iandobbie, 2020-04-21:
There appears to be 2 issues. 

1) Different shaped stages draw badly in the MacrostageXY panel

2) Objective offsets might cause the edges to be drawn off the visible regions of the panel.

carandraug, 2020-05-02:
> Different shaped stages draw badly in the MacrostageXY panel

Can you expand on this please? I'm not sure what different shaped stages you have and what badly drawn is for those (I haven't been to the macro stage window to be honest).

iandobbie, 2020-05-03:
Oh the default test stage, is similar tot he linkam stage in shape, horizontal and about 1:3 ratio in size. The DeepSIM stage is similar but in the opposite direction. These both work ok. The test stage I implemented in microscope is square and this was squashed non isotropically and left loads of empty space for no reason. Danny's edits have made it much better but not ideal see my comments in #587

carandraug, 2020-05-03:
Just to add more detail to the issue, here's some screenshoots with XY
stages of different sizes.

The default dummy XY stage with 25000x12000 dimensions:

[image]

A larger stage with 1:2 ratio (25000x50000):

[image]

A square stage (10000x10000):

[image]

A 1:5 ratio (10000x50000):

[image]

I have also found that the code does work with negative but things look weird.  Probably should error:

[image]

The images above were done by manually patching the `dummyXYStage.DummyMover` like so:

```
--- a/cockpit/devices/dummyXYStage.py
+++ b/cockpit/devices/dummyXYStage.py
@@ -59,10 +59,10 @@ import cockpit.handlers.stagePositioner
 
 class DummyMover(stage.StageDevice):
     def __init__(self, name="dummy XY stage", config={}):
-        config['primitives'] = \
-            "r 12500 6000 3000 3000\n" \
-            "c 5000 6000 3000\n"       \
-            "c 20000, 6000, 3000\n"
+        config['primitives'] = ''#\
+            # "r 12500 6000 3000 3000\n" \
+            # "c 5000 6000 3000\n"       \
+            # "c 20000, 6000, 3000\n"
 
         super(DummyMover, self).__init__(name, config)
         # List of 2 doubles indicating our X/Y position.
@@ -87,7 +87,7 @@ class DummyMover(stage.StageDevice):
     def getHandlers(self):
         result = []
         for axis, (minVal, maxVal) in enumerate(
-                [(0, 25000), (0, 12000)]):
+                [(0, 10000), (0, 10000)]):
             handler = cockpit.handlers.stagePositioner.PositionerHandler(
                 "%d dummy mover" % axis, "%d stage motion" % axis, True,
                 {'moveAbsolute': self.moveAbsolute,
```

carandraug, 2020-05-04:
@carandraug wrote:
> I have also found that the code does work with negative but things look weird. Probably should error:

scratch that, it was working fine, it was me who patched it wrong (although with negative numbers it looks really weird that the origin is on the bottom right corner).